prusa3d/PrusaSlicer #14459: Flatpak GNOME version 46 no longer supported

Frumious, 2025-04-18:
### Description of the bug

version 46 of GNOME is no longer supported and should be upgraded.

```
$ flatpak update
Looking for updates…

Info: runtime org.gnome.Platform branch 46 is end-of-life, with reason:
   The GNOME 46 runtime is no longer supported as of April 17, 2025. Please ask your application developer to migrate to a supported platform.
Info: applications using this runtime:
   com.prusa3d.PrusaSlicer

Nothing to do.
```

### Project file & How to reproduce

Run `flatpak update` on linux

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Most recent version of prusaslicer in flatpak

### Operating system

Ubuntu 25.04

### Printer model

Core One

SachCZ, 2025-04-18:
Hello, thank you, we are aware of the newer version, sadly there is an extremely obscure Nvidia driver issue, that we were not able to solve yet and it does not manifest on the version 46. It does manifest on version 47. If you look into the Flatpak history, you will see that we intentionally downgraded the platform. It is not a good solution (it is not even really a solution), but we do not have anything better yet.

TomCab, 2025-04-18:
> Hello, thank you, we are aware of the newer version, sadly there is an extremely obscure Nvidia driver issue, that we were not able to solve yet and it does not manifest on the version 46. It does [...]

---
Hello,
First of all, congratulations for the great work you are doing supporting Linux platforms.

Regarding the Nvidia issue you mentioned: Is there any chance that it has been fixed in runtime org.gnome.Platform branch 48? Minor version 48.1 was just released in April 2025 and fixed a lot of bugs, maybe your Nvidia issue does not pop up (although no Nvidia-specific fixes are mentioned in the release notes: https://download.gnome.org/teams/releng/48.1/NEWS)

shanefagan, 2025-04-22:
Could we maybe get a flathub-beta version of it for other graphics platforms? If there is an Nvidia issue some of us wouldn't be affected like with Intel or AMD graphics.

SachCZ, 2025-04-25:
Hello @MeanEYE, I would like to point out, that your comment is simply not true. How is it not working? As far as I know, the warning does not impact the functionality whatsoever.

See here:  https://github.com/prusa3d/PrusaSlicer/issues/14459#issuecomment-2814851693
The mentioned Nvidia **issue is not Flatpak specific**. If you do not use Flatpak it **may or may not manifest** based on your setup. The advantage of Flatpak is that we can have at least reasonable confidence in the fact that it wont happen with the platform version we tested.

In my opinion, this mentioned issue is much worse without Flatpak as it would manifest, but only sometimes, when the stats are aligned. **I think you chose the wrong thread to voice your opinion** as Flatpak in this case actually improves the situation.

CRCinAU, 2025-04-29:
Can we just get the AppImage back that doesn't have these issues?

SachCZ, 2025-04-29:
Hello @CRCinAU, to make sure your comment is not taken out of context, according to your own logic, there are 129 currently open Appimage problems: https://github.com/prusa3d/PrusaSlicer/issues?q=is%3Aissue%20state%3Aopen%20appimage

**But that logic is obviously flawed.** Please do not pollute threads with **genuine issues** with your complaints, it is of no use to anyone.

MeanEYE, 2025-04-29:
> Please do not pollute threads with **genuine issues** with your complaints, it is of no use to anyone.

AppImage no longer being provided is a genuine issue as not all of us want to use Flatpak. Regardless of number of problems each of those had it worked and people used it. In fact forcing us to use Flatpak is such a big issue for me, I am willing to stick with the version that works for as long as I can and then eventually switch if need be. For me Flatpak is simply too much.

SachCZ, 2025-04-29:
Hello @MeanEYE, I apologize for my wording. I did not mean to imply that no longer providing Appimage is not an issue for some people. What I meant is: Do not pollute **other** genuine issue threads. Use a dedicated thread, such as https://github.com/prusa3d/PrusaSlicer/issues/13653.

CRCinAU, 2025-04-29:
> Use a dedicated thread, such as [#13653](https://github.com/prusa3d/PrusaSlicer/issues/13653).

The realist in me just observes that this is a way to make the protest / request easier for Prusa to ignore.

myyc, 2025-05-09:
for what it's worth since this thread is filled with off topic flatpak hatred, i just wanted to add that i believe that flatpak is the best official universal choice for linux and i am happy (together with most users who care very little where their software technically comes from) that this choice was made, and i hope more distributors follow up. this bug is of course non-blocking and i hope that regardless prusa3d can soon get rid of the gnome46 dependency.

t1u1, 2025-05-12:
> there is an extremely obscure Nvidia driver issue, that we were not able to solve yet and it does not manifest on the version 46.

Is this issue documented somewhere? Can you please provide a link so that we can track it?

aaron-baff-ad-net, 2025-05-31:
Any update to this? Does the above mentioned org.gnome.Platform branch 48 fix the issue and so you can update to that and stop Flatpak complaining about the org.gnome.Platform 46 still being installed?

Drizzt321, 2025-05-31:
Any update to this? Does the above mentioned org.gnome.Platform branch 48 fix the issue and so you can update to that and stop Flatpak complaining about the org.gnome.Platform 46 still being installed?

Makalipay, 2025-05-31:
I have great respect for the folks at Debian but it's a huge bureaucratic organisation. Their package versions lag far behind. 

May i respectfully suggest switching to a distro with up to date repositories? 

For example Manjaro is as easy to install, use & maintain as any of the Debian / Ubuntu but is based on a well curated Arch eg packages are updated to stable Arch versions roughly every few weeks. 

No PPA's needed, no contortions needed.

Drizzt321, 2025-05-31:
@Makalipay not sure where Debian comes into this... but Flatpak is a different distribution mechanism and outside the scope of this ticket.

Makalipay, 2025-05-31:
> @Makalipay not sure where Debian comes into this... but Flatpak is a different distribution mechanism and outside the scope of this ticket. 

The warning is specific : the version of Flatpak in use by your distro is OUT OF DATE. 

The easiest solution is switch to a distro with UP TO DATE version of Flatpak. 

It couldn't be any clearer. 

Cheers

MeanEYE, 2025-06-01:
> > > [@Makalipay](https://github.com/Makalipay) not sure where Debian comes into this... but Flatpak is a different distribution mechanism and outside the scope of this ticket.
> > 
> > 
> > The warning is specific : the version of Flatpak in use by your distro is OUT OF DATE.
> > The easiest solution is switch to a distro with UP TO DATE version of Flatpak.
> > It couldn't be any clearer.
> > Cheers
> 
> Actually, incorrect. I'm getting PrusaSlicer from Flathub (https://flathub.org/apps/com.prusa3d.PrusaSlicer) via flatpak, and the Flathub flatpak of PrusaSlicer says "use org.gnome.Platform v46".
> 
> This is not a Debian vs other distro, this is the actual flatpak itself published by Prusa into Flathub using an outdated Flatpak runtime. As per the original ticket post, the flatpak application itself, coming from flathub, says "this is too old, tell the application developer to update their flatpak application".
> 
> ```
> $ flatpak list --app --columns=application,runtime
> Application ID                                                Runtime
> com.prusa3d.PrusaSlicer                                       org.gnome.Platform/x86_64/46
> ```

What @Makalipay is suggesting that `flatpak` application itself is outdate, not the flatpak package. But suggestion of switching distributions is not an option for me, especially not for this reason. Why not switch to Windows while we are at it, for compatibility reasons of course. Or even more productive, why not switch slicers. You can give a shot to SuperSlicer, BambooStudio, OrcaSlicer. All of which are based on PrusaSlicer and many if not all still offering AppImage packages.

What could be a possible solution is to pull flatpak from backports which is version 1.14. If that's not fresh enough to solve the issue perhaps try grabbing package from Sid, which is 1.16 version as opposed to 1.10 in stable.

cabbey, 2025-06-14:
> Just out of curiosity (and because I might be missing something), [@shanefagan](https://github.com/shanefagan), **as a user**, why do you actually care? It is not like the whole gnome "feel" of PrusaSlicer is going to be radically different. I understand why one might care from a technical point of view, but not the user one.

@SachCZ since I don't see this question has been answered, or if it was, it was lost in the sea of FUD, I'll give you a simple one: **as a user I do not want to be running unsupported software because there is no promise of fixing security issues in it.** After a while there are most likely going to be known security vulnerabilities in these older unsupported versions. As you add more online interactions into the slicer you open more avenues for possible vulnerabilities to be exploited.

Makalipay, 2025-06-24:
> > > > > [@Makalipay](https://github.com/Makalipay) not sure where Debian comes into this... but Flatpak is a different distribution mechanism and outside the scope of this ticket.
> > > > 
> > > > 
> > > > The warning is specific : the version of Flatpak in use by your distro is OUT OF DATE.
> > > > The easiest solution is switch to a distro with UP TO DATE version of Flatpak.
> > > > It couldn't be any clearer.
> > > > Cheers
> > > 
> > > 
> > > Actually, incorrect. I'm getting PrusaSlicer from Flathub (https://flathub.org/apps/com.prusa3d.PrusaSlicer) via flatpak, and the Flathub flatpak of PrusaSlicer says "use org.gnome.Platform v46".
> > > This is not a Debian vs other distro, this is the actual flatpak itself published by Prusa into Flathub using an outdated Flatpak runtime. As per the original ticket post, the flatpak application itself, coming from flathub, says "this is too old, tell the application developer to update their flatpak application".
> > > ```
> > > $ flatpak list --app --columns=application,runtime
> > > Application ID                                                Runtime
> > > com.prusa3d.PrusaSlicer                                       org.gnome.Platform/x86_64/46
> > > ```
> > 
> > 
> > What [@Makalipay](https://github.com/Makalipay) is suggesting that `flatpak` application itself is outdate, not the flatpak package. But suggestion of switching distributions is not an option for me, especially not for this reason. Why not switch to Windows while we are at it, for compatibility reasons of course. Or even more productive, why not switch slicers. You can give a shot to SuperSlicer, BambooStudio, OrcaSlicer. All of which are based on PrusaSlicer and many if not all still offering AppImage packages.
> > What could be a possible solution is to pull flatpak from backports which is version 1.14. If that's not fresh enough to solve the issue perhaps try grabbing package from Sid, which is 1.16 version as opposed to 1.10 in stable.
> 
> [@Makalipay](https://github.com/Makalipay) So I am running Debian/Testing (which is in freeze right now), and it most _DEFINITELY_ is _NOT_ a flatpak application version. Current version is 1.16.1 as per https://github.com/flatpak/flatpak/releases, and that is exactly what's installed.
> 
> As you can see, I have 1.16.1 of flaptak installed, and I'm _still_ getting the error. This is a Flatpak (from Flathub!) application configuration setting, specifically on Prusa. As explained above, the reason they aren't going from 46 -> 47 is that there's a weird Nvidia issue. Which while I don't run Nvidia, I do understand their reluctance to upgrade for the moment. However, since there's a v48 out, my question was to try and prompt Prusa to see if that fixes the issue with Nvidia, so they can use a newer and maintained version of the Flatpak Platform package.
> 
> ```
> $ dpkg -l | grep flatpak
> ii  flatpak                                             1.16.1-1                                 amd64        Application deployment framework for desktop apps
> ii  kde-config-flatpak                                  6.3.4-1                                  amd64        Flatpak permissions management KCM
> ii  libflatpak0:amd64                                   1.16.1-1                                 amd64        Application deployment framework for desktop apps (library)
> ii  plasma-discover-backend-flatpak                     6.3.4-3                                  amd64        Discover software management suite - Flatpak backend
> $ flatpak update
> Looking for updates…
> 
> Info: runtime org.gnome.Platform branch 46 is end-of-life, with reason:
>    The GNOME 46 runtime is no longer supported as of April 17, 2025. Please ask your application developer to migrate to a supported platform.
> Info: applications using this runtime:
>    com.prusa3d.PrusaSlicer
> 
> Nothing to do.
> $ cat /etc/debian_version 
> 13.0
> ```

Thanks Drizz321. A helpful clarification! 
*Small apology to anyone who thought I was singling out one particular distro.

sarowarhosen03, 2025-07-10:
> Hello. This thread got buried in a lot of noise, but it was never clarified if there is any motion to test the GNOME 48 runtime for example, or to fix the actual NVIDIA issue that is causing this (the issue itself was never specified either). [@SachCZ](https://github.com/SachCZ) was actively communicating here up until the end of April, then stopped, and never said anything more specific that a vague Nvidia issue mention. Can we get any clarification?

waiting for the upgrade

Drizzt321, 2025-07-10:
@eternal-sorrow thanks!

From my reading of that other issue, looks like there's a specific GLX render flag that turns on nvidia specific library, and whatever the conflict is that triggers it. So if that flag is NOT used, everything is fine? And so next release the Gnome platform will be moved to latest, with notes about that specific flag usage?

SachCZ, 2025-07-10:
Hello @Drizzt321, yes that is our current plan/understanding of the issue.

bit4man, 2025-08-14:
So 4 months in and still no resolution or even a preview/test image?

Frumious, 2025-09-15:
It looks like it's fixed now, and uses v48:

```
$ flatpak info com.prusa3d.PrusaSlicer

PrusaSlicer - The Slicer for Every Maker

          ID: com.prusa3d.PrusaSlicer
         Ref: app/com.prusa3d.PrusaSlicer/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 2.9.3
     License: AGPL-3.0-only
      Origin: flathub
  Collection: org.flathub.Stable
Installation: system
   Installed: 274.4 MB
     Runtime: org.gnome.Platform/x86_64/48
         Sdk: org.gnome.Sdk/x86_64/48

      Commit: 7f09d114bfebdf71b3aadffdda7e627866ad7cb9640bf169b834787a55780d52
      Parent: 58d6fbfc89f78bb1cbbb9150c0c346d88547dc56ed1a3f9723c9229004543066
     Subject: Release 2.9.3 (cce915551b46)
        Date: 2025-09-12 19:51:17 +0000
```