prusa3d/PrusaSlicer #14463: Option for paint on supports where negative volume touches an object

Slowmoe222, 2025-04-18:
[image]

I wanted to make a cutout inside a wall piece using a horizontal cylindrical negative volume so that I can insert a round multicoloured logo afterwards. My problem in the process is that the cylindrical cutout will most likely cause a print failure when there is a need to print the walll not face down but standing upwards. 
In this setting using the standart function to use supports everywhere causes the problem that small gaps created to interlock the wall parts will also get flooded with support material wich makes it almost impossible to create a smooth interlocking part in the end.

If an option to use paint on supports at spaces where a negative volume cuts through an object could be created I am pretty sure that would solve the problem I described.

Snuff1eupagus, 2025-04-19:
This is not an issue, Slicer can already accommodate this scenario. Simply add a support enforcer of a slightly larger size then the negative volume, e.g if your Negative Vol is 28.2 make the support enforcer 28.3 and slice. Paint on is a tool like any other, its not for every situation. 

[image]

This is more of a question for the forums then a suggestion for github I think.

Regards

u89djt, 2025-04-19:
@Slowmoe222 so far you've just got comments from fellow users like me.
The way negative volumes are represented and processed means that there isn't an explicit model surface to paint onto.
You could include an identical second copy of the negative volume, convert it to a support enforcer and just shift it upwards to intersect the final object.
There's another theoretical possibility that might make a more practical feature request: paint support enforcers onto the negative volume object. Converting the cylinder in this back to a negative volume and the block back to a part doesn't prompt supports to be created, but the support painting on the cylinder is retained if you change it back to a part again. So there's support info represented somewhere that might be persuaded to coincide with the extruded surface I suppose.
[image]
It was worth flagging the opportunity for more user-supporting functionality so it can be contemplated.

Slowmoe222, 2025-04-19:
> [@Slowmoe222](https://github.com/Slowmoe222) so far you've just got comments from fellow users like me. The way negative volumes are represented and processed means that there isn't an explicit model surface to paint onto. You could include an identical second copy of the negative volume, convert it to a support enforcer and just shift it upwards to intersect the final object. There's another theoretical possibility that might make a more practical feature request: paint support enforcers onto the negative volume object. Converting the cylinder in this back to a negative volume and the block back to a part doesn't prompt supports to be created, but the support painting on the cylinder is retained if you change it back to a part again. So there's support info represented somewhere that might be persuaded to coincide with the extruded surface I suppose. [image] It was worth flagging the opportunity for more user-supporting functionality so it can be contemplated. What I've suggested looks like a different way of reading your words :)

Since I got absolutely 0 developer skills I probably wasn't able to explain what I wanted perfectly on spot in the first place but your reading and translating of my request in your own words comes pretty damn close to  the vague mental picture I had in mind how the problem should be solved in future releases.
Although @Snuff1eupagus is certainly right when saying that a solution to my problem already exists, I personally think that if there is a more convenient way to solve a problem (and in my opinion being able to paint support enforcers on negative volume objects would be more convenient) this  approach should always be favoured to improve the overall user experience.

closed 2026-09-29 by prusaslicer-bot[bot]