prusa3d/PrusaSlicer #14464: Layer height of support doesn't match layer height of part with very thin layer setting

nesdnuma, 2025-04-19:
### Description of the bug

Hello,

I am printing a part in PLA with supports in PETG (using the Prusa XL 2T). The toolhead for PLA has a .1mm nozzle and the PETG toolhead has a .4mm nozzle.

After slicing I see that the layers of the supports (orange color in the picture) are much thicker than the set layer height (.05mm) (the part is in pink color in the picture).

[image]

### Project file & How to reproduce

[Test_buse 0_1_V8.zip](https://github.com/user-attachments/files/19820468/Test_buse.0_1_V8.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 10

### Printer model

Prusa XL 2T

nesdnuma, 2025-04-19:
Please note that with organic supports, the supports layer height is fine, but not with other support types.

neophyl, 2025-04-19:
Prusa slicer has always printed supports at different layer heights to those selected for the part.  Thats its normal default operation to save time and filament.  Unless you have the option to synchronise with object layers enabled.  Which is only selectable when the contact distance is set to 0.  

[image]

You don't have that option enabled in your project.

closed 2025-04-19 by rtyr