prusa3d/PrusaSlicer #14465: Incorrect solid fill

BillMcClennen, 2025-04-19:
### Description of the bug

In Print Settings>Horizontal shells>SolidLayers Set Top = 0, Bottom 3 (any number greater than 0).  After slicing **both** top and bottom are filled!  This is not just what is displayed, it prints that way also!  I am printing aircraft parts that need to be fit together and must have nothing in the top.

### Project file & How to reproduce

[CUB.zip](https://github.com/user-attachments/files/19821706/CUB.zip)

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

 Linux Mint 21.3 Cinnamon     Cinnamon Version 6.0.4

### Printer model

MK4S

u89djt, 2025-04-19:
(Community triage) The object you're slicing has cavities. What you're seeing at the top of the slice previdw is the bottom layer of the top solid region of the object being extruded as a single layer because you asked for a single bottom layer. The top of the last solid region of the object isn't filled because it's following the 0 solid layers instruction. That solid layer you're seeing isn't at the top. It's a bottom layer in the object.
[image]
You can close the issue yourself. Best wishes.

u89djt, 2025-04-19:
FYI, if you look in the 3mf file the authors provide, they're setting bottom and top layers to zero and using height modifiers to induce the base and cap of the whole objects. They're using the walls of the cavities to place two perimeters instead of the single perimeter specified for the object as a whole. They've put in some cunning technique to make a fast-printing lightweight object.

closed 2026-09-29 by prusaslicer-bot[bot]