prusa3d/PrusaSlicer #14468: Add Hole Horizontal Expansion

AltoidUC, 2025-04-19:
**Is your feature request related to a problem? Please describe.**
I'm requesting hole horizontal expansion as I need dimensional accuracy for the XYZ and holes. Other slicers such as Cura and Orca have a feature and it's very useful to me. Unfortunately I am not able to use PrusaSlicer as I need hole horizontal expansion for parts to go together smoothly.

**Describe the solution you'd like**
I'd like to see the feature added into the Windows, Mac and Linux build. It should be below the "XY Size Compensation" feature in "Print Settings > "Advanced" > "Slicing"

**Describe how it would work**
The feature would work similar to XY size compensation, but would have a focus on enlarging or shrinking holes specifically.

**Describe alternatives you've considered**
I've tried using the regular XY size compensation, but while XYZ are accurate, the holes are still too small.

**Additional context**
N/A

AltoidUC, 2025-04-25:
No responses 5 days later? This is a very useful feature and was hoping for some feedback on it.

neophyl, 2025-04-25:
Might be because its a duplicate of a VERY old feature request that has been re-requested multiple times.

For example https://github.com/prusa3d/PrusaSlicer/issues/4561  from 2020

closed 2026-09-29 by prusaslicer-bot[bot]