prusa3d/PrusaSlicer #14469: New Sequential Print Algorithm is a DISASTER, please remove it until ready for prime time!

sbierly, 2025-04-20:
### Description of the bug

Variations on this theme have been reported for sometime, evident in a couple dozen posted issues, focusing on various versions of the issue, such as the new algorithms is: nonoptimal, crashes, order is wrong, model for a specific printer is wrong, old manual guides were strangely taken away from us, etc. These are all true, big issues. Bottom line, there is only one fundamental issue: this very ambitious solution to a very hard problem is not remotely ready to be release, yet it's not only released, it's our only option short of regressing to 2.90, which many of us have done. 

Now, we have 2.92, and nothing is fixed. The many issues you know about here are generally dismissed, people are not even assigned to fix it. I doubt you can fix it any time soon, so instead, please either 1) Remove it completely, or 2) Add a checkbox to switch between the solution in 2.90, and that of 2.92+. 

Please consider this a priority, those of us that do sequential printing (nearly all I do) are greatly affected by this. 

### Project file & How to reproduce

[Broken Sequential Printing Example.zip](https://github.com/user-attachments/files/19823306/Broken.Sequential.Printing.Example.zip)

[image]

All I did is put some 25mm cubes on the plate, for the Mini, enabled sequential printing, and hit "arrange", and got this ridiculous mess which will cause both extruder and x-axis crashes. Parts are actually in the right order (in 2.9.1 they were not), but way too close, and almost arbitrarily placed. If I make them shorter/smaller (18mm), and add more, it caps out to 11 on a plate, still too close, with much of the plate unused.

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11

### Printer model

Printer Mini

u89djt, 2025-04-20:
(From a fellow user taking an interest)
There aren't any collisions in the example you've provided. Can you post a project with a crash caused by auto arrange? 
[image]

sbierly, 2025-04-20:
# @u89djt I humbly withdraw my issue as posted, and have edited it above with much less amplification. I further humbly admit that I had missed the amazing secret Prusa enhancement that shows the high resolution extruder when you move the horizontal slider after slicing...wow. I also applaud that the new autoarrange appears to do a full 3D analysis of the slicing based on the objects, and the extruder model, and has thus come up with very nonintuitive (at least, to me) sequencing compared to the rote "front to back, right to left" Mini procedure I've been using for years. Thus, for the case of my part in question, which only violates the collision height over a narrow x-region at the top, it figured out that it could place it much closer than I had previously believed and was doing, so the auto placement appeared crazy and wrong. It is a bit crazy, because it cares not about aesthetics, but only optimal placement, and is pretty awesome. 

I illustrated all this above as edited. Still, it is violating the prior 20mm height rule, and I can't check this as both my Mini's are printing long jobs right now. And it is unclear if the visualization is completely accurate. 

Now it is more clear why Prusa is resisting the old guide circles, because they are so inaccurate compared to this. Still, if we want to place parts manually, we need some better guides to do this. Short term, the circles (should it be an ellipse?), as before, are helpful, particularly if you give us knobs to set the size. Long term, it seems possible to render an 2D collision outline around all parts placed, and the current part being placed, to allow accurate manual placement, but I'm sure this is not trivial to implement.

u89djt, 2025-04-20:
You can place an stl of the mini's print gantry on the bed in the slicer and move it around if you want to investigate for yourself. The slicer will scale it down to fit but you can set it back to 100%. It needs cutting down so that a save and load will keep it as it is. The actual collision model is simpler: it's made of vertical rectangular prisms detailed in the json file, so it will miss some opportunities to be closer, but it's pretty good already.
There's always more to learn, and some of it has to replace what you thought you knew before. Heck of a ride, and there's a lot of people on it with you :)
C:\Program Files\Prusa3D\PrusaSlicer\resources\data\printer_gantries
[prusa3d_mini_gantry.zip](https://github.com/user-attachments/files/19826413/prusa3d_mini_gantry.zip)
[image]

sbierly, 2025-04-20:
Wow, another secret, tucked away on my own computer, thank you! This is a bit clumsy, but an excellent work around if you need to do a placement optimization. 

One question though. What happens when you have modded things? I replaced my stock hotend with this fantastic upgrade [](https://www.sliceengineering.com/products/prusa-mini-upgrade-kit). Shouldn't be much difference, but it could be 1-2mm difference in height, not sure. In that case, we are now missing (I think) any knob to offset for this in auto-arrange.

u89djt, 2025-04-20:
You'd have to edit the collision model in the json file in the printer_gantries directory to deal with it right now.
I picture a text box in the printer settings like they have for custom gcode. They'll have pictured more options :)

SachCZ, 2025-04-22:
@sbierly, can this be closed? I do not see a bug here.

sbierly, 2025-04-22:
Well, sure, I apologize for my initial misidentification of crashing
layouts, I learned something new :) But, the issue remains that for some
reason, despite the very accurate modeling now for Prusa Printers,
placement utilization is very poor, it chooses suboptimal ordering and
significantly under utilizes the plate. For the size parts in my example,
it might only place 4-5 on the Mini plate, where 8 can actually fit. This
is kind of a big shortcoming. As long as this is captured elsewhere, sure,
you can close this.

Thanks all.

u89djt, 2025-04-22:
If I can offer a perspective: the model of the gantry that you can see is very detailed, but so far, the collision model is more simplified. For the mini, it's a collection of eight vertical rectangular boxes.
Why? Path planning is a classic hard problem that takes experience and time to build efficient and robust methods for each new application.
They've started with a form of model - vertical boxes - and number of entities - eight for the mini - that balances fidelity of representation against time taken to compute a solution on common hardware.
The technology and results will evolve as they and the comminity develop the overall approach.
[image]

sbierly, 2025-04-22:
@u89djt Sage words, as always, many thanks for your perspective and information. That all makes perfect sense to explain the situation, but Prusa announced this as if it was big news and just works, not as a work in progress mentioning the inefficiency limits. I'm sure that over time they will evolve this into something much more optimal and usable. But, in the present, it's not really usable to anybody doing production printing, we can't print with ~60% efficiency on a plate, so the limitations really mean we are still placing by hand. That's why I, and many others, have been asking for the guide outlines (so far just circles) to be restored, and some option boxes for distance, to help us manually place quickly. Prusa has been indifferent, commenting that this is obsoleted by the new auto arrange---the important point is, that statement is premature, we still want those guides.

u89djt, 2025-04-22:
I think there's precisely zero chance that they're indifferent. They'll be brewing a solution for you. The apparently indifferent person was probably less involved in the technical side of the product.
Switching to resolving your need:
You could add a part to the object that shows the old school outline while you arrange them. Add modifiers to knock out perimeters, top and bottom and infill.
[outline.zip](https://github.com/user-attachments/files/19856179/outline.zip)
[image]

closed 2026-09-29 by prusaslicer-bot[bot]