prusa3d/PrusaSlicer #14471: too many travels

Spongman, 2025-04-20:
### Description of the bug

i'm printing the stringing test "Quick Stringing / Retraction Test" (https://www.thingiverse.com/thing:4893375)

[image]

i have basically the default settings.

the gcode i get when slicing does the following:

- new layer
- travel to right tower, fill right tower
- travel to left tower, fill left tower
- new layer ...

this results in twice the number of travels than are necessary.

it should be:
- new layer
- fill left tower
- travel to right tower, fill right tower
- new layer
- fill right tower
- travel to left tower, fill left tower
- new layer

it shouldn't do a travel immediately after moving to a new layer. it can just continue extruding where it is.

### Project file & How to reproduce

slice. view gcode

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

win 10

### Printer model

voron 2.4 with afterburner

u89djt, 2025-04-20:
(Fellow user taking an interest) You need to zip and post your 3mf project file. The settings you use can affect the extrusion and travel patterns significantly.

SachCZ, 2025-04-22:
Hello, did you take cooling into account when constructing your argument?

Spongman, 2025-04-22:
[Quick_Stringing_Test.zip](https://github.com/user-attachments/files/19858507/Quick_Stringing_Test.zip)

completely vanilla setup. i couldn't get any different behavior with any other settings.

u89djt, 2025-04-23:
It still does it when the layer time would be sufficient for cooling to take place, I think?
[800pc_Stringing_Test.zip](https://github.com/user-attachments/files/19863738/800pc_Stringing_Test.zip)
[image]
So there is an opportunity to be more efficient in general in extrusion planning. This is a known opportunity - it's a simpler aspect of partial completion of regions of objects when there's room for the gantry. Think of sequential printing with effectively lots of partial objects on top of each other.
For the specific example of the stringing test, as @SachCZ intimated, you'd expect cooling to be a problem: your towers would be very melty. It's a stringing test, so you want to see the strings to be able to judge how you're doing.
It would be good to be able to tell the slicer to stay as local as possible at any given stage of the print.