prusa3d/PrusaSlicer #14472: Blob after layers above 4th in Vase Mode

Sarecto, 2025-04-20:
### Description of the bug

PrusaSlicer 2.9.2, Printer Creality Ender S1 with Klipper

After update of Slicer trying to print a flower pot in vase mode, and I figure out a problem.
Slicer adding for all layer above 4, some extrusion and trashing my print. I checked all settings and didnt find anything. 
Look at the pictures, screens of gcode and maybe u can find some issue. Before update all was clear and perfect.

[image]

[image]
[image]

### Project file & How to reproduce

[tornado_pot_6h33m_0,15mm_220C_PETG.zip](https://github.com/user-attachments/files/19824739/tornado_pot_6h33m_0.15mm_220C_PETG.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

win 10

### Printer model

Ender S1 Klipper

u89djt, 2025-04-20:
(Fellow user taking an interest ) You've posted the gcode, but you actually need to post the 3mf project file.

Sarecto, 2025-04-20:
I havent 3mf, i have stl

[tornado_pot.zip](https://github.com/user-attachments/files/19824838/tornado_pot.zip)

u89djt, 2025-04-20:
You need to use the "save project as" button in the slicer to create a 3mf file that contains all your settings and objects for the project. Save it just before you slice it and it gives that result you've shown a photograph of.

Sarecto, 2025-04-20:
Oh, ok i thought u need model file :D here you are.

[tornado_pot.zip](https://github.com/user-attachments/files/19825024/tornado_pot.zip)

u89djt, 2025-04-20:
OK, so remember that I'm a fellow user, not one of the devs.
You're using an S1 profile from somewhere else.

Switch to the ALPHA Creality Ender 3-S1 profile provided in the printer settings wizard (other FFF -> creality) and see if that works. 

ALPHA means that it's being tested experiementally as an inclusion with the slicer and they're not sure about it yet. That's a profile provided by people outside Prusa. If you still get a blob and no-one here finds a flaw in your file or the slicer when you're using the ALPHA profile (zip and post that), you need to find a way of interacting with the authors of the profile, which doesn't happen in this repository.

General comments:
Your blob you've shown is at the start of the vase regime. The first four layers are printed as distinct solid layers.
You've set your filament diameter to 1.74 instead of the default 1.75. That will lead to a higher extruder feed rate, which will only be correct if your filament is actually under-sized. I can't see how that would lead to a blob in just one spot, though.

Sarecto, 2025-04-20:
Yes, I use other profiles, but these are original profiles from Prusa modified by me. They have been working since I got the printer and nothing has changed in them for 3 years. I use different nozzles, different materials and different manufacturers of filaments so I had to create my own configurations and if the update did not change something I would not look for the fault on my side. Several dozen kilograms of materials went through this printer without major reservations and suddenly such a problem.

u89djt, 2025-04-20:
Have you printed in vase mode before?

Sarecto, 2025-04-20:
Dozen times and many materials.

u89djt, 2025-04-20:
What are they doing differently to you in the current ALPHA profiles?

Sarecto, 2025-04-20:
As I said, I use Klipper, so start codes, end codes, my coordinates, thumbnail preview gcode, machine limits. Nothing that has any bearing on what the actual problem is at the moment. Besides, I've been using these profiles since the beginning when Prusa wasn't yet supporting this printer model. Thanks a lot for your involvement, but we still have no clue what's wrong.

u89djt, 2025-04-20:
Some hints from three years ago: https://github.com/supermerill/SuperSlicer/issues/3245

Sarecto, 2025-04-20:
I have enabled this allways. Thanks for halping, but it doesnt fix it.

Sarecto, 2025-04-20:
In my opinion, something happened with the filament function "retraction when changing layers" because checking or unchecking it does not force re-cutting. I think that such things occurring thousands of times in the print have a key impact on the code, since the slicer does not require cutting, something is not working. I deleted all E14 values ​​in the notepad after the 4th layer and I am starting the print. We will see tomorrow what changes.

Anticipating further questions, I reinstalled, changed profiles, changed Prusa versions. I've been working on one print for two days now and I've used almost half a roll for dozens of ruined prints.

[image]

u89djt, 2025-04-21:
You've overriden deretraction length to 14, which is huge, and not altered retraction length to compensate. Hence the big blob of plastic.
[image]

Sarecto, 2025-04-21:
As I said, I've never had this problem, I've printed a few kilograms of this filament and it was ok. This is the specificity of my equipment and I've printed on these settings without a problem and I haven't changed anything. Whenever I buy something new I do all the tests such as flow rate, diameter deviation, input shaper etc. it worked on my printer. now it doesn't.

u89djt, 2025-04-21:
In fact I got it entirely the wrong way up - I highlighted retraction length, and you've set the deretraction length in filament overrides. That G1 E14 F1800 drives the extruder 14mm. So the retraction is being omitted? Nope, it seems to be happening all over the place. I don't understand this:
[image]
Is it something to do with the steps in the model? The slicers's having to place seams at various points up the model despite being in spiral vase mode. 
[image]
A cylinder doesn't show any seams.
[image]
Has that same model worked ok in a previous version of the slicer?
All the G1 E-14 and G1 E14 commands pair up in the gcode you posted originally and in what I generate now, so there's no extra plastic being requested. Is it failing to retract?

u89djt, 2025-04-21:
OK, so the paired E-14 and E14 commands are when I use the ALPHA profile.
With your profile, there are unpaired E14 commands which of course emit extra plastic.

u89djt, 2025-04-21:
I didn't notice that you added this text, sorry:
"I tested other models and after analyzing the gcode I came to the conclusion that the problem does not occur in every layer. This is a case of several layers and a dozen or so normal prints. Then again several layers with a blob and again correct printing. This extrusion "e14" is the retraction value that I have assigned to this filament. If I change it to even 8, the blob still occurs but is smaller. Something is wrong with the software here. I have no idea what went wrong and I hope it is a matter of some patch.

The problem also occurs when I turn off vase mode and reduce the number of strokes to 1."

So, something in the difference between your 3-year-old profile and the ALPHA profile causes the E-14 to be missed in 2.9.2.
There will be information in that difference that helps in the bug hunt. I don't know if you'll be able to see everything exposed in the slicer - diffing the profile files seems like the next step.

u89djt, 2025-04-22:
Ok, so the seams are added because "avoid crossing perimeters" is set in your 3mf.
That seems like a bug to me. I suppose avoid crossing perimeters should be irrelevant during the main body of spiral extrusions? It can be switched off as a workaround.
The retraction and deretraction demands in the gcode with and without avoidance add up to zero, so no extra plastic is being demanded in the gcode.
In the original avoiding case, we see only G1 E-14 and G1 E14 pairs in the spiral region, but incremental retractions during the solid first layers. You're not getting blobs in those solid first layers, so the incremental retractions are successful (enough).
You get the blobs where the retraction is a single G1 E-14 command.
That suggests that your printer is failing to execute a retraction of 14.

u89djt, 2025-04-22:
There a big single retraction command at the end of the brim - is there any kind of glitch in the brim there? Just noting here that progressive retractions aren't being highlighted as such:
[image]

SachCZ, 2025-04-22:
Hello, I must admit I find it very hard to follow your conversation. Can you please summarize? Is there a bug with vase mode and avoid crossing perimeters or is it just misconfiguration?

u89djt, 2025-04-22:
Hi there, I did a really bad job this time.
When avoid crossing perimeters is active while slicing this model in vase mode, the resulting slice has a number of seams scattered along the surface. That in itself doesn't seem to be placing extra extrusion commands in the gcode. I think the print should succeed.
I think those seams shouldn't be generated, because it's within the spiral vase zone.
It seems like seams are being inherited by the spiralization code? I haven't checked the source.

But the print doesn't succeed where the G1 E14 command immediately follows a G1 E-14 command: it results in a blob on the issue author's print as seen in the photo.
Where a G1 E14 follows a number of incremental retractions in a wipe pattern, it doesn't seem to create a blob.

These seem to be fine:
[image]

These seem to result in a blob:
[image]

That might suggest that retraction is failing on the printer when it's a fast, long E move.

SachCZ, 2025-04-22:
Thank you, I agree. There the seams all over the place seem to be a visual bug and should no actually affect the gcode. The huge retraction and deretraction is probably a misconfiguration on @Sarecto side. My suspicion is that this might not have manifested previously because the generate gcode was slightly different and the travel might not have been there.

Summary:
- there is a visual bug
- the 3mf is misconfigured

I have created an internal issue for the visual bug: SPE-2754.

Thank you for your time and feel free to disagree if my conclusions are incorrect.

u89djt, 2025-04-22:
I didn't think of it being a visual bug - that's a clue. If you're indicating a seam where there's a full deretraction, that might say something.
There are unbalanced G1 E14 commands. Here's one:
[image]
That doesn't happen with the ALPHA profile, but it shouldn't happen at all, so I guess that's your bug: not a visual seam bug, a spurious deretraction bug.
Is it being copied from the planar slice into the spiral?

u89djt, 2025-04-22:
If you don't get confused during a bug hunt, you're not trying hard enough :)

Sarecto, 2025-04-23:
Thank you very much "u89djt
" for your persistence and pursuit of the goal. I think you may be right that it has something to do with retraction. Every dozen or so layers there are a few with uncontrolled extrusion. As if it wanted to fill something or replace the virtual withdrawal of the filament. My test print with the file modified in notepad failed. Somewhere I deleted too many of these errors which ended with deleting also good retraction and the filament fell out of the extruder. I did not make any more attempts. I only looked through the gcode. It did not occur to me that there could be a problem with the seam...

u89djt, 2025-04-23:
You're very welcome - I learned stuff too.
I think it could be worth looking at the most recent profiles provided with the slicer. They're built by a community, so they might have found solutions you hadn't noticed yet.

closed 2026-09-29 by prusaslicer-bot[bot]