prusa3d/PrusaSlicer #14475: Wipe while retracting has significant problems on stock MK3S+

Wegerich, 2025-04-21:
### Description of the bug

For a long time I have been having print quality problems where the start of each layer is missing material. It is most obvious on thin boxes, which I print often

[image]
[image]

After spending some time learning how to review GCode I have found some problems with the "wipe while retracting" feature.

[image]

1. The wipe feature has been messing up my prints as seen in the pictures above. If I disable it, my print problems go away. I don't know what the benefit of this feature is, but I would suggest making it disabled by default.
2. Indeed, the tooltip for "wipe" suggests that it is supposed to be disabled by default. 
4. I wanted to use "de-retraction extra length" to try and troubleshoot this issue, but the "wipe while retracting feature" silently blocks "de-retraction extra length" from doing anything. Therefore I thought over-retraction wasn't the problem, since changing de-retraction didn't fix it!
5. The "wipe" GCode generated completely ignores the retraction speed specified in Prusa Slicer (25mm/s? no you get F8640 (144mm/s) instead!)

### Project file & How to reproduce

[Tiny box for testing printing error (2).zip](https://github.com/user-attachments/files/19828124/Tiny.box.for.testing.printing.error.2.zip)

With these settings (wipe disabled), I get this GCode

[image]
[image]

Note the retraction at 3022 (E-.6) and subsequent de-retraction after moving at line 3028, including the 0.1mm extra. 

If I change the settings just by enabling the wipe feature, you can see the lines 3165-3171 controlling the wipe. The Feed rate is set to F8640 and then the complex wipe movement is carried out. 
After the movement, we see the de-retraction on line 3178 has been reset to 0.6mm (ignoring the 0.1mm extra). Still, the de-retraction speed is set to F1200, nicely aligned with the 20mm/s specified in the configs.

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11 23H2 22631.5189

### Printer model

MK3S+ stock, unmodified

Wegerich, 2025-04-21:
After some poking around revealed that changing `Travel for non print moves` gave a proportional change in the feedrate for the wipe extrusion, I looked for something that could be multiplying by 48 or 0.8 to give the 6840 number but didn't get anywhere. I guess it's not something users can change. Hopefully this is a good lead for investigation

SachCZ, 2025-04-22:
Hello,

1. seems to be a bug
2. it depends on what profile are you using
3. I got lost in the various speeds. Please help me here. What do you suggest exactly, is that the feedrate during the wipe is too high? There is a difference between the retraction speed and the feedrate for the whole wipe movement. If the retraction speed is lowered the wipe gets longer to keep the feedrate constant.

Thank you.

SachCZ, 2025-04-22:
1. Thank, I have created an internal issue for the bug: SPE-2755. Just a fair word of warning, we will not get to this in the near future.
2. This is intended. Whether that is a good choice I do not know.
3. We will look into lowering the speed.

Wegerich, 2025-04-22:
1. Thanks. It says in the tooltip that not many people use this, indeed the only time I wanted to use it is to troubleshoot the print quality problem in my photos.
2. If the wipe feature works well, it seems like a great idea. However, it seems to have a very negative impact on print quality for me. I suspect that it is a problem with the way the wipe feature is implemented, but it will require some investigation to understand exactly why this problem happens, and the main troubleshooting step is by enabling `deretraction extra length`. It does not happen with the same retraction speed, same retraction distance but no wipe. I wonder if the wipe movement is causing sticky filament to get "pulled" out of the nozzle during the wipe, therefore meaning that when the filament de-retracts, there is actually some empty space in the nozzle.
3. Thanks. I think there is a discussion to be had on this. Clearly my prints are being ruined and there are two fixes - slowing retraction speed or disabling wipe. Reducing retraction speed to 10mm/s slows a benchy from 52m to 56m on a MK3S+, so it's not trivial, but should we sacrifice that speed when not everyone experiences this problem?

Wegerich, 2026-09-05:
Anyone else looking at this can see my workaround tips here: https://github.com/prusa3d/PrusaSlicer/issues/11914#issuecomment-5549251956