prusa3d/PrusaSlicer #14482: show color change time already in slicer

swa1977, 2025-04-22:
Hi, 

I am wondering why the slicer does not show how much time it takes before the color change? Would be sometimes handy to know the time when the filament has to be changed for colour change, like when leaving the house at certain time and then could optimise the amount of items to be printed so it does not wait for filament change when I am away already. Now it shows only total print time in normal or stealth mode, but not when the filament change is going to happen, I only see that from the printer after starting the print. 

[image]

neophyl, 2025-04-22:
It does.  Hover your mouse over the slider where the colour changes are 

[image]
as you can see from the example its at about 22 minutes and 51 seconds.

Also on the preview if you change the legend to colour you get -

[image]

swa1977, 2025-04-22:
ahh thank you!

closed 2025-04-22 by swa1977