prusa3d/PrusaSlicer #14484: Please add proxy capability

hman2, 2025-04-23:
**Is your feature request related to a problem? Please describe.**
Slicer always "greets" me with the error message that it cannot check for updates for itself and the configuration. This is because our strict firewall rules prevent that applications "phone home". Communication through a proxy server, however, would be granted.

**Describe the solution you'd like**
Introduction of the capability that Slicer communicates with Prusa via a proxy server, and a GUI interface so I can specify the IP address of our proxy server.

**Describe how it would work**
Standard proxy interface and a GUI to manually set the IP address to use.

**Describe alternatives you've considered**
There is no alternative. And exemption from the company rules is possible, but not free of charge and has to be well reasoned. A "this add does know how to use a proxy" is not accepted from our security department.

hman2, 2025-04-23:
As requested in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4621

hman2, 2026-01-08:
The problem has now worsened!

Starting with 2.9.4 Slicer will always (!) try to download the configuration manifest (which btw. is called "mainfest") from Prusa, which must fail (not requested properly through the proxy).

And to make matters even worse, the workaround that comes to mind, set Slicer to load the manifest locally does not work. The manifest is available offline, and Slicer is configured to load that, but STILL it tries to run against our firewalls... :-(

For the latter I will open a bug ticket.

ggbce, 2026-09-18:
If Slicer 3.x support Internet proxy settings this ticket can be closed in my side.