prusa3d/PrusaSlicer #14495: Feature request - Cooling on support interface material

AltoidUC, 2025-04-25:
**Is your feature request related to a problem? Please describe.**
The issue is that sometimes supports can be challenging to remove, and a fan option for support interface material would allow the supports to come off in a much less rough matter.

**Describe the solution you'd like**
I'd like to see a feature added to "Filaments" > "Cooling" > "Fan settings" called "Support interface fan speed"

**Describe how it would work**
This feature would work by checking if the current filament being printed is related to support interface material, and will adjust the fan speed based on the setting.

**Describe alternatives you've considered**
I've attempted using 100% bridges fan speed and keeping my regular fan speed partially high, but I prefer to keep in relatively in the middle.

**Additional context**
N/A

Tupson444, 2026-09-19:
still wanted

Jan-Soustruznik, 2026-10-23:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16167.

closed 2026-10-23 by Jan-Soustruznik

Tupson444, 2026-10-27:
> 7\. A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion [#16167](https://github.com/prusa3d/PrusaSlicer/discussions/16167).

This request is not the same as Fan modifier at layer (that would need to be manually set and would affect the whole layer), instead we want to be able to set fan speed for all and only support interface automatically.

Nbeknel, 2026-10-28:
In theory this is possible to do in PrusaSlicer, though I haven't tested it in the 3.0-beta version.
What you do is create a copy of your filament profile and call it, for example, "PETG Support interface", set the parameter `fan_always_on` to true and change general cooling parameters to your requirements.
Say, we assign the standard profile to the first nozzle and the supports profile to the second nozzle. In the print preset in the multiple extruders tab, set `support_material_interface_extruder` to 2 and `support_material_extruder` to either 1 (no cooling) or 2 (cooling) (if 0, you'll have support material that is cooled in some layers and isn't in other). Then go to the output parameters tab, add a G-code substitution, select regex: Find - `\n(T\d+)`, replace with - `\n;${1} - comment out tool change`. This'll comment outall tool change commands.
If you're using a multitool or multimaterial printer: Find - `\nTi`, replace with - `\nTj`, where i is the index of the extruder with the supports profile and j the extruder with the base profile (e.g. Find - `\nT5`, replace with - `\nT0`).

For example, this an ABS part with no cooling, but 50% for support material interface:
[image]