prusa3d/PrusaSlicer #14498: Snapmaker 2.0 Unknown command M73

jalapenopuzzle, 2025-04-26:
### Description of the bug

I am using a Snapmaker 2.0 A350T printer.
My print logs contain many messages similar to the following:

```
Recv: echo:Unknown command: "M73 P22 R20"
```

The Snapmaker 2.0 G-Code reference at
https://wiki.snapmaker.com/en/Snapmaker_Luban/manual/2_supported_gcode_references
does not document `M73`.

I see from
https://marlinfw.org/docs/gcode/M073.html
that M73 is supposed to be provided by Marlin firmware, but Snapmaker Marlin has left it out.

In this test, I saved the .gcode file to a USB stick and plugged it in to the printer.
I observed the output using Octoprint. (The job was not sent using Octoprint in this case).

I have not tested sending the job by Octoprint yet, but I don't see how that can be relevant. The printer documentation says it does not support M73, and the log shows that the printer complains about it.

The printer still prints, so this is just an annoyance.

### Project file & How to reproduce

[cube.PrusaSlicer.3mf.zip](https://github.com/user-attachments/files/19920477/cube.PrusaSlicer.3mf.zip)
This is just a flow rate calibration cube taken from Teaching Tech's printer calibration website 
<https://teachingtechyt.github.io/calibration.html#flow>.

GCode sliced from the project file provided above.
This GCode contains `M73` progress updates.
[cube.PrusaSlicer_0.16mm_27m.gcode.zip](https://github.com/user-attachments/files/19920495/cube.PrusaSlicer_0.16mm_27m.gcode.zip)

This is a (partial) log file of the printer output while printing the gcode file above.
This log file includes the printer complaining that it doesn't recognise `M73`.
[cube.print.log](https://github.com/user-attachments/files/19920474/cube.print.log)

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Ubuntu 24.04

### Printer model

Snapmaker 2.0 A350T Dual QS BK

neophyl, 2025-04-26:
M73 is inserted because your snapmaker profile is configured to add them.  Under Printers>General>Firmware you have an option called Supports remaining times ticked.  

[image]

Untick this and it wont add the M73 lines to the gcode output anymore.  Once unticked save the modified profile and then use that to slice with in the future.

jalapenopuzzle, 2025-04-26:
I have checked my printer profiles.
This box is not checked on my Snapmaker 2.0 profile, but it is checked on the Octoprint Snapmaker 2.0 profile.
I didn't enable this box on either printer profile.
I created the Octoprint printer yesterday.
I observed the problem while printing from the printer directly from a USB key. I believe the print was generated using the Snapmaker 2.0 profile, before I created the Octoprint profile. I'll have to double check all of this.

neophyl, 2025-04-26:
The project you attached has the option ticked.   That screen shot is directly from your project.

rtyr, 2025-04-28:
Remaining times were just disabled via configuration update for all Snapmaker printers. It should resolve this issue.

closed 2025-04-28 by rtyr