prusa3d/PrusaSlicer #14499: Feature request - Upstream Sovol SV06 Plus ACE

GameTec-live, 2025-04-26:
**Is your feature request related to a problem? Please describe.**
Orcaslicer has a sovol SV06 Plus ACE profile included, but i (very much) prefer PrusaSlicer.
Orcaslicer also has a way more limited selection of filament profiles (atleast none are visible when selecting the SV06 Plus Ace)

**Describe the solution you'd like**
Itd be great if the Sovol SV06 Plus ACE profiles could be upstreamed and the full filament selection is available.

**Describe how it would work**
It looks like Orcaslicer (and Bambuslicer) use a different way of storing profiles (as a .json file), so someone would probably have to recreate those profiles in a prusaslicer compatible way?

**Describe alternatives you've considered**
Use Orcaslicer, but, atleast in my opinion, Orcaslicer is a ruined version of PrusaSlicer

TerPuc, 2025-04-28:
Hello @GameTec-live, third party printer profiles are created by community. If the printer profile isn’t implemented in PrusaSlicer, you can create the profile yourself or add it as a config bundle. However, I recommend visiting the forum for Sovol users and asking for advice there. 

Keep in mind that creating or importing a printer configuration requires some additional skills, as an incompatible profile could potentially damage your printer. So, proceed with caution I suggest using the software recommended by the printer manufacturer.

closed 2025-04-28 by TerPuc

GameTec-live, 2025-04-28:
Thanks for the response.
As Orcaslicer is just a prusaslicer fork and other sovol printers are already supported, i thought there might be some internal tool or similar to just easily convert formats.
So i thought I'd just ask.

If i port over the profiles, what would be the path of submitting them into prusaslicer? Just doing a PR?