prusa3d/PrusaSlicer #14504: Bridge algorithm fails

NielschB, 2025-04-28:
### Description of the bug

The bridge algorithm seems to fail on attached model while it should be fairly straightforward to print this in a correct way. It mostly selects the printing start point exactly at the bridge start which makes it fail in most cases. Beside this the remaining bridge paths that are selected make most of the bridging fail.

[image]

[image]

[image]

### Project file & How to reproduce

[Bridges_fail.zip](https://github.com/user-attachments/files/19947517/Bridges_fail.zip)

Please see attached project file, slice 0.2mm and check layer 17

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2 win64

### Operating system

Windows 11  24H2

### Printer model

Custom build based on mk3s

neophyl, 2025-04-28:
This is a duplicate of a very old issue, see https://github.com/prusa3d/PrusaSlicer/issues/9074  (and others)

NielschB, 2025-04-28:
I see, bit surprised that it's not fixed when it is know for such long time.
Should I close this ticket?

u89djt, 2025-04-29:
(From another user taking an interest) It looks quite different to #9074 because the bridges aren't started in mid-air. The extrusion look to me like they should generally succeed except for the apex formed /into/mid air, and the curved bridges, which will of course droop because they're not held taut in the correct shape.
[image]
The classic perimeter generator looks like it might turn out better:
[image]
The principal issue seems to be that you need curved bridges, and they can't work well, perhaps unless you go extremely slowly during that layer. You might just have to enable supports under the curved bridges.
What does that layer look like in the physical plastic result?

NielschB, 2025-04-29:
The curving is only minor and I would accept that the bridges result in straight lines. The problem with the starting points is that the filament has only a very tiny connection surface on bridge start. There are other tickets on bridging perimeter overlap which was insufficient at some point, also causing bridges to fail while one theoretically would expect that it should succeed. I assume its due to mechanical stress caused by shrinkage of the filament when cooling down just after it got extruded.
Fully agree on using supports for better results, however that's a workaround and does not solve this issue.

I think the algorithm should just not start so close near the edge of the bridge locations.

When I change the extrusion width just a bit, at some point it even decides to print bridge mid air as indicated below, but I understand that this is a known issue.

[image]

u89djt, 2025-04-29:
The example project from #9074 still extrudes bridges into thin air. That's a bug.
You should probably post the 3mf file of your example where you got a mid-air extrusion.
[image]

Tupson444, 2026-09-19:
still relevant