prusa3d/PrusaSlicer #14509: Wipe tower on XL:: size too large, purge volumes cannot be changed

smurfix, 2025-05-01:
### Description of the bug

Enabling the wipe tower on the XL results in a tower that's excessively large given that we only want to get rid of a bit of ooze / undo the retraction when parking the tool.

The docs say "The purge amount can be set by clicking on the Purging volumes... button in the right panel."

The code says `wiping_dialog_button->Show(wipe_tower_enabled && multiple_extruders && single_extruder_multi_material);`

However since the XL doesn't have a MMU (for very good reasons) this check fails and the wipe volumes cannot be changed.

### Project file & How to reproduce

See above.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

all

### Printer model

XL multi-head

rtyr, 2025-05-02:
Filament settings --> Advanced --> Minimal purge on wipe tower

This value allows to set how much volume is purged on the wipe tower for non-single-nozzle setup.

smurfix, 2025-05-06:
ah, thanks.