prusa3d/PrusaSlicer #14511: slowdown_below_layer_time is wrongly int, should be float

richfelker, 2025-05-02:
### Description of the bug

Self-explanatory. It's impossible to set fine-grained min layer times because 2s is way too long, 1s is often too short, and some printers want even below 1s such as 0.8s or 0.5s.

### Project file & How to reproduce

NA

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

All

### Operating system

All

### Printer model

All

richfelker, 2025-05-02:
Yes, you can type a value with a fractional part, but it won't be honored.

u89djt, 2025-05-02:
Ah, here?
[image]

u89djt, 2025-05-02:
It seems to be float everywhere else, so it might just be that one slip in PrintConfig.cpp

richfelker, 2025-05-02:
The type is also wrong in PrintConfig.hpp, which makes it take forever to test the fix because everything has to be recompiled. I'm testing it now...

richfelker, 2026-09-02:
This issue is still present, and even has an open PR that should be easy to rebase to fix conflicts if anyone is willing to consider it for merging.