prusa3d/PrusaSlicer #14512: Support paint being ignored with modifier

Ilovemyhous, 2025-05-02:
### Description of the bug

Hello!
I'm trying to print the model provided in the attachment, however, I did wanted that the model had 100% infill after a certain layer height, so I used a modifier, set it's scale and size, along with the infill for the mesh/model inside the modifier.
But here's the problem, after setting-up the modifier, it ignores my red support paint.

I'm aware that I could put the entire model at 100% infill, or set the supports to "For support enforcers only", but maybe someone has the same issue as be, but with a different model.

This is the support structure with the modifier (that just modifies the infill).
[image]

Compared to the one without the modifier.
[image]

And this is the support paint.
[image]

### Project file & How to reproduce

https://github.com/user-attachments/assets/3063d0b3-0055-42a8-80a7-b6d98d724e6d

[Crown GitHub.zip](https://github.com/user-attachments/files/20018158/Crown.GitHub.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Windows 11 24H2 (Build 26100.3775)

### Printer model

Original Prusa i3 MK3S

u89djt, 2025-05-02:
Ah, but if you knock the global infill down (I picked 20% at random), your problem appears. That would appear to be a bug.
[image]

Ilovemyhous, 2025-05-02:
Whops, yeah I set the global infill to 100% so that I didn't need to rely on the modifier, forgot to change it back down for the issue. ^^'
Good catch!

u89djt, 2025-05-02:
Using a height range modifier instead of a shape modifier gives the same problem. 
[image]
Separate bodies are generated during the slicing process for regions with different infill percentages, so I guess it loses track of the paint.
I was about to suggest that a potential workaround would be to split the object to parts at that height and set infill and paint, but that exhibits exactly the same problem.

neophyl, 2025-05-02:
How about this as a workaround to get it printed.  

[Crown GitHub_neo.zip](https://github.com/user-attachments/files/20019182/Crown.GitHub_neo.zip)

[image]

Its one reason why I generally use enforcers rather than blockers, you get more control.  I also sort of flipped the modifier to cover the bottom (which didn't help with the original bug anyway).  Also used all perimeters on the upper bit rather than 100% infill, should print faster and better.

Ilovemyhous, 2026-09-04:
Tried the 3.0.0-alpha11 slicer, and while it respects now the 100% for the modifier, and x% outside the modifier, it still doesn't apply the support paint with the modifier. Removing the modifier does apply the support paint successfully.