prusa3d/PrusaSlicer #14518: soit disant trop grand

18pool49, 2025-05-04:
### Description of the bug

une alarme nous prévient que le STL est top grand ou que tout est pas sur le lit ???????

[image]
et cela avec la totalité des STL (petits ou grands a très grands) 

### Project file & How to reproduce

il y a pas besoin de commencer le tranchage l'alarme arrive immédiatement 

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

W11

### Printer model

XL5 et toutes autres machines

neophyl, 2025-05-04:
Given that you have attached no project file or even an stl of what you are trying to load for anyone to duplicate the issue its going to be a bit hard to track any problem down.

18pool49, 2025-05-04:
tout les STL produisent cette alarme au moment ou il arrive sur lit 

[v2-stl-vague-by-art3dchoix.zip](https://github.com/user-attachments/files/20028750/v2-stl-vague-by-art3dchoix.zip)

TerPuc, 2025-05-05:
Hi @18pool49, we are unable to reproduce the issue with the information provided. Please provide the STL file and describe the exact steps that led you to this issue. 
Also provide a screenshot with the sidebar on the right side.
Thanks

18pool49, 2025-05-08:
voilà une vidéo du déroulement de la venue de cette alarme qui se produit sur et a l'ouverture de tout les STL sans exception

18pool49, 2025-05-08:
https://github.com/user-attachments/assets/fc125130-95ae-4ff8-b714-30f11a24b98f
voilà le "zip" de ce STL >>

[V2 STL Vague By Art3Dchoix.zip](https://github.com/user-attachments/files/20098408/V2.STL.Vague.By.Art3Dchoix.zip)
voilà les 2 prises d'écran du cartouche de droite comme demandé 

[image]
[image]

TerPuc, 2025-05-14:
Hi @18pool49, thanks for the video. I needed 3mf file which contains settings of your project😊because I am not able to reproduce the issue only with the stl. So I wanted to take a look at the project. But I see you closed the issue. Is it solved?

18pool49, 2025-05-14:
non je n'ai rien résolu ; 
je suis navré ,c'est plus un raz le bol que j'ai , et je vais faire avec !!  
car je ne trouve pas  d'origines cohérentes  , tout les STL ou presque tous , car j'en ai trouvé un qui me produit pas cette alarme , >> histoire de pimenter la situation et l'incompréhension 
alors vous laissez tomber ma question  ,étant donné que ca ne complique pas la suite de la programmation 
salutations

TerPuc, 2025-05-15:
Hello @18pool49, I understand. We are not prioritizing this in this case at the moment. But I will keep it in mind and check if any of the STL files I use behave similarly. Thank you for your support! 😊

closed 2026-09-29 by prusaslicer-bot[bot]