prusa3d/PrusaSlicer #14520: (XL-multitool)   P/S assigns a wrong tool for support.

S3D-r, 2025-05-04:
### Description of the bug

I aim to print everything with **T3**, except the support _interface_ _layers_ with **T5** (for easier support removal, different material). However, is a certain layer includes support interface, then it will all the support in that layer with _Support Interface Tool_ (=T5).

### Project file & How to reproduce

[image]

[KupuZIP.zip](https://github.com/user-attachments/files/20029764/KupuZIP.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

W10-64

### Printer model

XL-5T

neophyl, 2025-05-04:
I dont have an XL but I think its because you have left the Support material extruder defined as 0.  I think thats auto so its using the same extruder for the rest of the support on the same layers to reduce filament changes.
If you Set the Support Material Extruder to the same as the other material, ie 3 in your case you get what you want.

[image]

It even says that's how it works on the tool tip 

[image]

S3D-r, 2025-05-06:
Thanks! So, a user fault...

kubispe1, 2025-05-07:
Thanks for the help. From the conversation, it's seems to me this is a setup issue concerning the support_material_interface_extruder, not a bug. 
We agree the naming is long and could be clearer. I am closing this issue but we may need to think how to rename it.

closed 2025-05-07 by kubispe1