prusa3d/PrusaSlicer #14521: Bridges with no support not being treated as bridges when soluble support is enabled

pipcorona, 2025-05-05:
### Description of the bug

When printing an object using soluble support settings and support enforcers, bridges that don't have support are not treated as such and instead are printed the same speed as solid infill material, instead of bridge infill.

This is the model the layer before the bridge. On the layer with the bridge, I pause and then insert some square nuts into the part.
[image]

This is the expected result, and what happens if detachable support is used. The spot directly above the nuts is printed as bridge infill.
[image]

This is the result if soluble infill is used, even though there is not support material under this bridge.
[image]

This results in the area above the nut being printed too fast and sometimes failing.

### Project file & How to reproduce

Use an object that has bridges or overhangs
Turn on soluble support for enforces only
Paint on some supports, leave other bridges without supports
Slice the file

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

macOS Sequoia 15.5

### Printer model

Prusa XL

neophyl, 2025-05-05:
https://github.com/prusa3d/PrusaSlicer/issues/6868#issuecomment-919949654  
basically when soluble support is activated it turns off ALL bridging.

pipcorona, 2025-05-05:
> [#6868 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/6868#issuecomment-919949654) basically when soluble support is activated it turns off ALL bridging.

It seems like it should only turn it off where there is support. So maybe this should be a feature request instead of a bug?

neophyl, 2025-05-05:
Given that it is specifically designed to work the way it does now, then yeah any change would be a feature request and not a bug.
Also given the age of that linked issue and the several times it has been reported since then good luck with that :)

hugo-id, 2025-08-27:
Just want to flag I ran into this issue today as I didn't realise the preset I was using had soluble supports.

In my case, selecting anything other than "none" for supports meant that the slicer didn't detect any bridges or overhangs anywhere.

This is a major UIX problem as with the MMU I can be printing one part with soluble supports, but numerous other parts don't have any supports at all and need overhang management.

Ie, my current part uses soluble supports, but just above it is a bridge that would print fine if soluble supports didn't break all bridging.

closed 2026-09-29 by prusaslicer-bot[bot]