prusa3d/PrusaSlicer #14523: Additional Support layers on top of the Dense Support (S Interface) layers.

S3D-r, 2025-05-05:
### Description of the bug

Additional Support layers are built on the top of the Dense Support (S Interface) layers. They serve absolutely nothing, the need of _Support_ (naturally) ends on the uppermost _Dense Support_ layer...

### Project file & How to reproduce

[image]

[Kupu 1-d.zip](https://github.com/user-attachments/files/20034126/Kupu.1-d.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

W10-64

### Printer model

XL-5T

u89djt, 2025-05-05:
(From a fellow user)
You've set a 10mm XY offset, so everything is pushed outward a long way.
The support interface deals with horizontal overhangs. The object continues above that overhang.
You've set a 20mm pattern spacing, so the squares are huge, and if any part of a square lies under a horizontal overhang, it's given a support interface over the whole square.
It's doing what you asked it to do.
[image]

S3D-r, 2025-05-06:
The 10 mm was my mistake, supposed to be 10%...

I understand why the support structure extends horizontally so much from the supported part of the item. But that wasn't my issue. The issue is those extra support layers on top of the dense support. They are meaningless as there's no dense support on top of these extra layers, and these extra layers just don't do anything.

u89djt, 2025-05-06:
They are there to support the curved region after the horizontal flat region, but the spacing you set puts them too far away to do so.

S3D-r, 2025-05-06:
> They are there to support the curved region after the horizontal flat region, but the spacing you set puts them too far away to do so.

Oh, now I see...

closed 2025-05-12 by Jan-Soustruznik