prusa3d/PrusaSlicer #14524: No warning in case of missing support

senicram, 2025-05-05:
### Description of the bug

I marked where the support should be done and 'force support in marked places' but still I have a place where missing support can make a lot of troubles for user and maybe nozzle can be in dust do to printing in air. In my opinion user should be warn about it.

[image]

[image]

### Project file & How to reproduce

[bearing_my.3mf.zip](https://github.com/user-attachments/files/20034457/bearing_my.3mf.zip)

I uploaded my project.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 10

### Printer model

MK4S

u89djt, 2025-05-05:
(From a fellow user) Consider what would happen if that exposed ring wasn't there. If the slicer warned of unsupported regions, it would complain about the parts of the surface you've partially supported. In choosing "For support enforcers only", you're taking responsibility for support, and I think the assumption has to be that you'll be examining all the blue regions for yourself. It would always complain, I think.

senicram, 2025-05-06:
In case if user is beginner it would be helpful to warn user. I made support for one place but forget to put support on ring and I expect warning from Slicer. Warning won't block printing but only warn user. We are only humans and we can forget or just miss a bug but Slicer shouldn't.

senicram, 2025-05-06:
OK, understood.

Another issue is about generating nozzle movements regarding the same model which is included into this issue. Please consider modifying the slicer to generate print paths that go across the support lines, especially when printing bridges. This would significantly reduce the occurrence of sagging, as the nozzle would be moving perpendicular to the direction of the supports, allowing the bridge to be better supported during printing.

Currently, I had to manually adjust the supports twice to match the nozzle's movement direction. This took extra time and could have been avoided if the slicer automatically aligned bridge paths across the supports. It would make the slicing process more efficient and improve print quality for overhangs and bridging scenarios.

u89djt, 2025-05-06:
I'm just another user trying to save the devs time dealing with things that aren't bugs or omissions. :)
The bridging is extruded to maximize likelihood of success according to the object definition. You'll see that bridges aren't structured differently according to presence or absence of support (switch them off and see there's no change). It's created before supports are generated. The sequence doesn't allow for iteration between object extrusions and support extrusions. It couldn't be implemented in the current overall scheme (correct me if I'm wrong, folks) but it's not theoretically impossible, and I imagine it's on a long list somewhere :)
When you take responsibility for supports, it's entirely your opportunity/problem to do cunning things that rely on an overall comprehension of the scenario that the slicer can't form.

Jan-Soustruznik, 2025-05-12:
Hello @senicram, Thank you for your report. We will investigate it more thoroughly. I’ve created an internal issue (SPE-2764).
Regarding the second reported issue, please create a separate ticket for it. Thank you, and have a great day!

closed 2026-09-29 by prusaslicer-bot[bot]