prusa3d/PrusaSlicer #14526: Layer shift with very high density polygon STL

gaellafond, 2025-05-05:
### Description of the bug

Slicing very high density polygon shape (from STL file) result in invalid GCode instructions (I believe it's an incredibly fast move in Y axis, much faster than the printer can do), which result in a loud "bang" and a layer shift.

**NOTE**: I sliced the shape twice, printed it 3 times, with the same result, at the same layer height.

**Workaround**: Right-click the shape in PrusaSlicer and select "simplify model". The STL file went from 4MB to 244kB without visible difference.

### Project file & How to reproduce

Project file: [gear-reduction_gear_layer-shift.zip](https://github.com/user-attachments/files/20036398/gear-reduction_gear_layer-shift.zip)

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.9.2+flathub.org

### Operating system

Ubuntu 24.04.2 LTS

### Printer model

Prusa MINI+

u89djt, 2025-05-05:
(Fellow user taking an interest)
With the information you've given, the leading hypothesis would be that the tips of the gearing have pulled up during extrusion and cooling of the big solid top layer leading to the nozzle hitting one of them and causing a layer shift.
I think your mini can't acclerate hard enough of its own volition to create a bang, but the deceleration due to a collision will bang and lose steps.
If you can specify what layer height the bang happens at and include a photograph of the print, someone might be able to come up with something else.

gaellafond, 2025-05-06:
> (Fellow user taking an interest) With the information you've given, the leading hypothesis would be that the tips of the gearing have pulled up during extrusion and cooling of the big solid top [...]

Thanks for trying to help, but I seriously doubt that's the issue. It always fails at the same layer (Z = 3.35), right after the layer change. The nozzle is in the centre of the part when it makes that bang, and the layer shift occurs, far from the teeth of the cog. The nozzle is not catching on anything. After I simplified the model, the problem went away. I managed to print the gear several times.

It might not be because of the density of the polygons. There might be something wrong in the geometry that PrusaSlicer can't fix when loading the STL. It's hard to analyse.

I will upload a photo, if that can help.

u89djt, 2025-05-06:
OK, post the 3mf file of the simplified version that worked...

gaellafond, 2025-05-06:
Here is a .3mf file, containing the simplified gear.
[gear-reduction_gear_simplified.zip](https://github.com/user-attachments/files/20060452/gear-reduction_gear_simplified.zip)

u89djt, 2025-05-06:
What do the undersides of the failed and successful prints look like?
The simplified model has slightly softened transitions in the tooth profile on the bed, which may have kept adhesion when the sharper original failed. That would allow more lift due to shrinkage during cooling.
We don't have a conclusive explanation yet. There's more information to gather.
[image]

u89djt, 2025-05-06:
You've picked a different first solid layer infill speed. That would change adhesion.

u89djt, 2025-05-06:
[image]
[image]

u89djt, 2025-05-06:
This is github. All the rare and unlikely behaviours land here. The folk reporting them can naturally feel singled out.

gaellafond, 2025-05-06:
The underside or all the prints looks good. There was no issue with bed adhesion...

It's hard to photograph, especially at night. They are perfectly flat, they didn't started to lift at any point.
[image]

It's not an issue with the print, it's definitely an issue with the GCode. The bang always happens at the exact same point. I was next to the printer for the last 2, I saw it happened right in front of me:
- The layer 3.20 finishes, while the nozzle is inside the part (finishes the last inner layer).
- The nozzle lift to start the next layer.
- The bed moves very rapidly to the centre of the part, for a split of a second, loud bang, layer shift
- The printer start printing the layer 3.35, starting with the circle in the centre.

It's like if the instruction to move the nozzle from the perimeter to the centre is set with very high velocity. It tries to teleport to the centre of the part. All other layers are fine, but that one is wrong. The GCode instruction is invalid. I don't know what causes the slicer to generate that invalid instruction.

u89djt, 2025-05-06:
When a part warps, parts of it rise slightly. That's what causes the vast majority of tool crashes. The photo seems to show separation on the failed parts - peeled regions - when the successful parts look smooth. It could be a trick of the light, but can you examine them and compare?
Chasing the gcode hypothesis: can you locate the place in the gcode where the move occurs? Drag the bottom slider to expose the gcode.

u89djt, 2025-05-06:
The command structure appears to be identical. Can you see any difference?
[image]
[image]

gaellafond, 2025-05-06:
Could this be the problem?
```
G2 X77.269 Y101.797 I.064 J.04 E.00505
G1 X77.649 Y101.873 E.0101
M204 P2000
;LAYER_CHANGE
;Z:3.35
;HEIGHT:0.15
;BEFORE_LAYER_CHANGE
G92 E0.0
;3.35
M201 X3993.92 Y3993.92
G1 E-2.5 F4200
G1 Z3.35 F720
;AFTER_LAYER_CHANGE
;3.35
M74 W2.64164

G1 Z3.55
G1 X86.41 Y88.311 F24000
G1 Z3.35 F720
G1 E2.5 F2400
```

A feedrate of 24'000 mm/min seems excessive. But all layers have that instruction, I'm not sure that layer in particular causes issue.

gaellafond, 2025-05-06:
> peeled regions

There are no peeling regions

gaellafond, 2025-05-06:
It's getting late here, I need to go to bed. I will run that bad gcode again tomorrow with no plastic filament, and I will film the bang. Hopefully you will believe me when I say the nozzle is not catching on anything.

gaellafond, 2025-05-06:
I have a Prusa MK2 and Prusa MK3. My printers are in storage (transitioning between locations). I got a Prusa MINI+ while transitioning. I have been 3D printing for many years. I know what you are talking about, that's definitely not the case here.

gaellafond, 2025-05-06:
Sorry, I think I'm getting tired (and a bit impatient). Good point about the nozzle temperature, I will manually adjust the GCode before trying that. I don't want to fry the heat block, PTFE tube, etc.

u89djt, 2025-05-06:
No worries, I'm happy to supply patience.

gaellafond, 2025-05-08:
Thanks @u89djt 

I have edited the GCode file. I changed M104 and M109 to 40 C. When I started the print, the printer gave me a warning saying the GCode was made for a different type of filament. Not sure how it detected that, but that was nice to see. Confirmed that it was correct and went with the print.

I started recording at Z=3.20. It got to Z=3.35 without the bang. I assume I was wrong. I was about to stop the recording, but decided to let it run. It turns out the layer shift happens between Z=3.35 and Z=3.50. And yes, it did bang and layer shift, even though there was no filament at all.

I'm not sure we can hear the bang in the recording. I will review the footage and upload to YouTube. I don't think there is anything useful in there, but the layer shift does happen even with no filament.

Now that I know the layer shift happens between Z=3.35 and Z=3.50, I will go back to the GCode and see if I can spot anything weird.

All the prints so far:

- **1st print**: I was away from the printer, I didn't see when the layer shift happened (about 2mm on Y axis). I measure the Z height with the callipers and it's at the same height as the other failed prints.
- **2nd print**: I was right in front of the printer when it happened. I heard the bang, saw the layer shift (about 2mm on Y axis), cancel the print.
- **3rd print**: I re-sliced the gear, with a 15° rotation. At the end of the layer Z=3.35, it did a much louder bang, and much larger layer shift (about 6mm on Y axis). I cancelled the print.
- **4th print**: I re-printed the 1st GCode, using my hacked version of GCode without filament, at 40°C. At the end of the layer Z=3.35, it did the bang and layer shift (probably 2mm on Y axis, I can't measure). It was less loud than I remember (because the last I remembered was the 6mm layer shift). I stopped the print.

gaellafond, 2025-05-08:
We can hear the "bang" at the end of the video. The sound is not very loud, you might need headphone. It sound more like a "Tack" than a "Bang" in the video. It's not quite the same that I heard while filming, but that should be good enough.

Video: [TODO - ADD AFTER UPLOAD]

u89djt, 2025-05-08:
We can note that if it is substantially quieter than the observation with a printed outcome, it might be an incidental operational vibration maximum coinciding with a contradicting direction change. We don't know yet. If you find a gcode command that could sensibly cause it, then of course we maintain interest. If not, then it is more likely that we need to look at the print in more detail?

gaellafond, 2025-05-08:
> It bangs loudly with filament and tacks quietly without it.

No. It bang loudly with 6mm layer shift (when I rotate the part 15 degree before slicing).

It's less lour with the 2mm layer shift (slicing without changing anything).

u89djt, 2025-05-08:
Yep, we need an export to chime in.

gaellafond, 2025-05-08:
I had a look at the GCode at the beginning of Z=3.35 and compared to the GCode at the beginning of Z=3.50.
I can't see anything that would explain the printer's behaviour, but I'm far from a GCode expert... I think someone with more experience should have a look at it.

Maybe I could copy a small fragment of GCode and paste it in a file and run it on the printer. I could reduce it until I find the one instruction (or combination of instructions) that cause the issue. Then compare with the equivalent in the 6mm layer shift to confirm that the instruction is also there, but more intense. That will be a difficult job, and I don't think that will be good for the printer.

u89djt, 2025-05-08:
Yeah, if we're moving to potential printer-damaging territory, I shouldn't have an opinion.

gaellafond, 2025-05-08:
I guess I will have to learn GCode...
I will start with a header home the printer head, then move it in the centre (x, y and z), before executing the fragment of GCode.

I hope I will be able to isolate the issue quickly, I don't want to damage my printer...

gaellafond, 2025-05-08:
Thanks for you help @u89djt , and thanks for you patience!

I will wait for the dev's input. If I find another part that triggers the bug, I will add it to this issue.

I believe it's an acceleration thing. It moves at "F24000" between layers, which is fine when the movement can be share between the X and Y axis, but not when it's all done by the Y axis. It's a wild speculation. If I'm right, we would see that issue come up much more often.

Another idea could be related to high polygon density. A lot of GCode instructions for very small movements. The printer gets overwhelmed, skip instructions (I don't even know if that possible) and end up going from one position to another way faster than it can move. Again, a wild speculation and that one is even harder to test. Also, I would not expect that one to be as reliable as it is: always at the exact same layer height, always the same layer shift...

u89djt, 2025-05-08:
If you look at the gcode size for the simplified and unsimplified cases, they're pretty much identical. The overall quantity of instructions will relate more closely to the slice and gcode resolution than the mesh complexity unless the mesh is very coarse and hence requires less than the resolution settings.

Jan-Soustruznik, 2025-05-13:
Hello @gaellafond,

Thank you for your report. I printed the model you provided, and all prints were completed successfully on our MINI test farm without any issues. - I used your print settings without any modification.

I also reviewed the G-code, and everything appears to be in order. However, I noticed that your start G-code includes a check for firmware version 6.1.2, which is outdated. The current firmware version is 6.2.4. Could you please update your firmware to the latest version? (I recommend it.) You can find it here: https://help.prusa3d.com/cs/article/aktualizace-firmwaru-mini-mini_124784 Additionally, I suggest checking the printer hardware — ensure the belts are tension and verify that the gear wheel rotates correctly during bed movement.  https://help.prusa3d.com/article/adjusting-belt-tension-mini-mini_157324

I hope that it you help you. Thank you, have a nice day

gaellafond, 2025-05-14:
Thank you @Jan-Soustruznik 

I didn't know there was a new version of the firmware. I will update it and give it another try.

Maybe I should also re-create my custom Printer / Print settings. I made a copy of the original a little while ago (with an older version of Prusa Slicer), to save the settings I like (save GCode as text, so I can inspect it, and use cubic infill). I suspect the GCode that check the firmware version is somewhere in there.

closed 2026-09-29 by prusaslicer-bot[bot]