prusa3d/PrusaSlicer #14527: Plater view rendering OpenGL issue on Wayland with 570.144 Nvidia Driver

Ionshiv, 2025-05-05:
### Description of the bug

I'm getting rendering failure on the plater view in prusa slicer using wayland and the 570.144 Nvidia driver. I'm futhermore using EndeavourOS. I'm far from sure this is an PrusaSlicer issue, but since it was using PrusaSlicer I ran into this I figured I'd make an issue.

When running the slicer in debug mode I'm getting 
OpenGL DEBUG message [Error/API/high]:
`(1286): GL_INVALID_FRAMEBUFFER_OPERATION error generated. Operation is not valid because a bound framebuffer is not framebuffer complete.`

### Project file & How to reproduce

This happens when booting the software. Just a black screen in the plater view. All other menus work as intended. 

[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

EndeavourOS - latest as of 2025-05-05

### Printer model

MK3.9

svenstaro, 2025-05-06:
Oh, thankfully I'm not the only one. I'm getting the same on Arch Linux.

HenryJia, 2025-05-06:
Also having this issue, on Arch Linux, any suggestions anyone?

svenstaro, 2025-05-06:
The flatpak works for me so it's probably a bug in our (Arch Linux) packaging :/ Please report a bug in our tracker.

svenstaro, 2025-05-06:
I fixed the issue downstream so please close this issue @Ionshiv :)

closed 2025-05-06 by Ionshiv