prusa3d/PrusaSlicer #14529: Wipe tower is unuseable w/ small Z heights: improvement ideas

smurfix, 2025-05-06:
The wipe tower rips off its base too easily because its attachment to the brim is just one single strip of extruded filament around its perimeter. As the tool head enters the tower at the same position every go-around, oozed blobs accumulate until the tower's attachment to the base fails.

On my multitool XL the effect of this problem is that I am categorically unable to print anything multi-tool-ish with <0.25mm layer height. The tower *always* rips off the base mid-print.

A good way to fix this IMHO would be a setting to add a chamfer around the base of the wipe tower, so it's attached more strongly and thus survives when a stray blob of filament blocks the print head.

Also, based on what I see in my print it'd help a **lot** for the head to enter the wipe tower at some random position, so that blobs that do get dragged around on the print head won't accumulate all in the same place.

smurfix, 2025-05-06:
Also one might want to print more than one wall around the tower, and/or set a minimum width/length

u89djt, 2025-05-08:
Would there be any mileage in running two independent towers of twice the layer height and alternating between them?

smurfix, 2025-05-08:
> Would there be any mileage in running two independent towers of twice the layer height and alternating between them?

That might work too, though I'd want to combine it with my random-start-position fixup. Time to start hacking the slicer …

closed 2026-09-29 by prusaslicer-bot[bot]