prusa3d/PrusaSlicer #14530: Different material for the uppermost Dense Support layer only.

S3D-r, 2025-05-06:
XL-5T. I have a big print that needs plenty of support. Therefore I use loose support (20 mm line distance). I also want easy removal (PETG part, supports tend to stick) and good surface quality, so my choise is different material between support and part,. Due to loose support network, I use four Dense Support (later: D/S) layers. I can choose different material for the D/S, and I use PLA for it . The best option would be printing the D/S with same material than the part, to achieve good bond between D/S and Support. For easy removal of Suppoert, only the very uppermost D/S layer would be printed with secondary material (PLA in this case). But that option is missing. Now the only option is to print all the D/S with the secondary material.  That turned to result in a weak merge between D/S and Support, causing Support to fail.

Jan-Soustruznik, 2025-05-15:
Hello @S3D-r,
I think that you can use  the **Support material/raft interface extruder** feature, located (_Print Settings > Multiple Extruders_) to select a secondary material to the connection support/model surface. Additionally, example you can set the **Contact Z distance** to **0 mm (soluble)** in the Support Material settings, and adjust the **Top Interface Layers** to control the number of secondary material layers between the support and the model surface.

I hope I understood your requirements correctly and that my suggestion helps.

S3D-r, 2025-05-15:
Hi! Thanks for your proposal, I try to be clearer with my need...

1. I want the _basic support_ to be same material than the part in order to minimize tool changes.
2. I want the Dense Support (=Interface Layers) to be same material than the _basic support_, in order to get a good bond between the basic support and Dense Support. 
3. I want the **_uppermost_** Dense Support layer (I have multiple Dense Support layers) to be different material in order to be able to use zero Z gap and still easy removal.

Your suggestions don't fulfil my needs.

Jan-Soustruznik, 2025-05-19:
Hello, @S3D-r,  thank you. I understand. This feature is not currently implemented. Did you try my proposed solution? I mean, adding this feature isn’t just about providing an additional option — it would require extensive testing on our printer farm. From what I see, my solution could achieve the same results. Do you think there would be any significant difference?

npburns224, 2025-09-05:
This seems to be closer to how Bambu Studio approaches multi-material supports (see issue #12062).

I’ve been experimenting with printing models and wipe towers in PLA while using PETG for the support interface. The problem I often run into is that the first PETG interface layer doesn’t adhere well to the sparse PLA support structure below it. With so little surface area—and poor bonding between the two plastics—it often fails to form a solid connection.

Bambu Studio solves this differently:
1. The main support structure is printed in PLA.
2. Just before the interface layers, an additional solid PLA layer is printed as if it were the first interface layer.
3. The PETG interface layers are then printed on top of this solid PLA layer, which provides a flat, continuous surface with far more contact area, making adhesion much more reliable.

When printing parts in PLA with both the support structure and support interface in PETG, adhesion is significantly more reliable because of the increased surface area. However, this requires a tool change on every layer, which substantially increases print time.

To avoid that, I’ve been manually modeling an “interface layer” in PLA for all of my parts to mimic Bambu Studio’s approach. With this method, PETG adheres ***much*** more reliably while avoiding the excessive tool changes. It would be very helpful if PrusaSlicer could adopt this behavior natively to simplify the workflow.

S3D-r, 2025-09-05:
> Hello, [@S3D-r](https://github.com/S3D-r), thank you. I understand. This feature is not currently implemented. Did you try my proposed solution? I mean, adding this feature isn’t just about [...]

I didn't try, as it doesn't help how many dense support layers you have, if the dense support doesn't stick to the sparse support. Which is the case when you have different dense support material compared to sparse support material.

nickpburns, 2026-09-23:
This feature is still absent and would be a huge improvement to the reliability to multi-material supports.