prusa3d/PrusaSlicer #14531: Misplaced support material

smurfix, 2025-05-06:
### Description of the bug

This screenshot of a print with soluble support interface shows random strips of support material between the support interface and the actual object to be printed.

That should not happen.
 
[image]

### Project file & How to reproduce

[Shields.zip](https://github.com/user-attachments/files/20061416/Shields.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Linux

### Printer model

XL multi-head

u89djt, 2025-05-06:
(Community triage effort)
The zip file there is actually a renamed 3mf file. It contains two objects, one of which is set to not print. That hides a separated sliver in the object that is being sliced.
Here a zip file containing a 3mf file of just the object with the original settings being printed so the whole thing is visible.
I'm not seeing any overlap of support and object being interleaved.
[SingleShield.zip](https://github.com/user-attachments/files/20061551/SingleShield.zip)
[image]
[image]

smurfix, 2025-05-06:
@u89djt I just opened your file with my copy of PrusaSlicer; the problem is reproducible:

[image]

NB

u89djt, 2025-05-06:
You're seeing this fragment, which is part of the support structure and not interleaved with the object. It's there because of local interactions between the overall support structure and requirement for support interfacing under the object.
[image]

u89djt, 2025-05-06:
It's absolutely deterministic, resulting from an algorithm that makes sure support material remains coherent in a wide range of complex and counterintuitive circumstances, but it is unnecessary in that very specific spot on your object.

u89djt, 2025-05-06:
Just to tie it up: Was this plastic dropping onto the object what motivated a search for a misbehaviour?
[image]
You can close the issue yourself if you're sorted.

smurfix, 2025-05-07:
@Snuff1eupagus Huh. Would you mind (a) telling the rest of us which part of what @u89djt is factually incorrect (and preferably correct them …), (b) toning down the aggressive language somewhat?

closed 2026-09-29 by prusaslicer-bot[bot]