prusa3d/PrusaSlicer #14533: Fatal error "Access violation" on slicing multimaterial/multicolor prints

protostheticspope, 2025-05-07:
### Description of the bug

[image]

Looks like this was an issue before and it was fixed, maybe part of the code got reverted? 

### Project file & How to reproduce

[error slicing.zip](https://github.com/user-attachments/files/20091572/error.slicing.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11

### Printer model

RatRig Vcore 4

protostheticspope, 2025-05-07:
Further info - The error only happens when I generate automatic supports and have a raft enabled. If I disable either of these, the error doesn't come up.

protostheticspope, 2025-05-07:
Further info #2 - The error does not happen if I use the multimaterial painting tool to paint part of the model.

TerPuc, 2025-05-12:
Hello @protostheticspope, thank you very much for your report and investigation. Internal issue SPE 2763 has been created.

lukasmatena, 2026-09-01:
Fixed in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena