prusa3d/PrusaSlicer #14537: Resizing / rescaling multipart models feature

KlausKraemer, 2025-05-08:
Sometime one would like to resize only part of a model. For example a cylinder with threaded ends and an unthreaded middle part. If you need another length of this rod, it would be convenient to only resize the middle part and leave the threads as they are. Otherwise the threads would be outside the specification.

[image]

If you increase / decrease the height of the red middle part, this happens:

[image]
[image]

The now longer middle part 'swallows' up or disconnects the ends. This is very unfortunate, because many times only one of multiple parts would have to be changed in its dimension to get a desired result.

There should be a way to keep the order of parts in a multiple part model and place them in the original order. Especially for those, downloading models with no CAD knowledge, this would be very helpful.

neophyl, 2025-05-08:
This is basically a duplicate ofhttps://github.com/prusa3d/PrusaSlicer/issues/7563

KlausKraemer, 2025-05-08:
> This is basically a duplicate of [#7563](https://github.com/prusa3d/PrusaSlicer/issues/7563)

Yeah, it has its similarities. Before creating the thread I searched for 'resize' and 'rescale' but came up with nothing.

Anyway, I believe, some makers without construction skills would benefit a lot from such a feature and especially models downloaded from sources like Printables could be manipulated easier.

closed 2026-09-29 by prusaslicer-bot[bot]