prusa3d/PrusaSlicer #14545: with some 3MF files objects drop to the floor without asking

RainerMtb, 2025-05-11:
### Description of the bug

[image]
[image]
[X1.zip](https://github.com/user-attachments/files/20147637/X1.zip)

With most 3MF files containing multiple objects we get a dialog asking how the objects should be handled. This does not happen with the example file included. Also see the discussion in the forum https://forum.prusa3d.com/forum/prusaslicer/multiple-objects-in-3mf-file/

This seems to be a controversial topic, Prusa Support does not see this as a bug. I consider it a bug because this behavior is unpredictable as some files load correctly while with some files all objects drop to the floor. I cannot find a problem on the side of creating the 3MF file as my example loads fine in other software. My example was created in latest version of Fusion 360.

### Project file & How to reproduce

Open the 3MF file attached in PrusaSlicer 2.9.2 and observe that all objects drop to the floor without asking. Compare to other files containing multiple objects where a dialog "multiple objects..." shows up when importing

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11

### Printer model

MK4S

u89djt, 2025-05-11:
(A fellow user taking an interest)
The stl import function asks about grouping parts, but the 3mf import doesn't do that. A 3mf file is intended to assert the object relationships and the slicer respects that during import. The thread and cap are independent objects, so they're treated that way. 
@neophyl indicates the area https://forum.prusa3d.com/forum/prusaslicer/multiple-objects-in-3mf-file/#post-740160 

There is a workaround that hands the objects importable from the 3mf directly into a group.
Open your 3mf file and note that suspended objects have dropped.
Select one of the objects, right click, "add part->...load"
[image]
And select the 3mf file. The objects are imported as parts of the parent you selected.
[image]
From there you can duplicate, delete parts and you've got what you need.
The answer, as neophyl stated, is to save as a mesh file, not a 3mf file
Why not just check anyway? Parts are ascribed project settings in a 3mf file that need to be associated with the right groupings.
You'd be asking for the code to take a detour for special cases where a project file contains no settings and no structural information.
Just save as a mesh.

u89djt, 2025-05-11:
Obviously you could just add a box to the bed and add part from loaded file to save some deleting.

u89djt, 2025-05-11:
OK, there's a better answer. Group the parts in the source package before exporting as 3mf. That can preserve independent objects with multiple parts. If you just saved as stl and loaded as a single part, you'd have to do all the copying and deleting.
Here's your original 3mf imported into Fusion with the head and thread merged preserving mesh faces, duplicated to demonstrate the value of this route.
[groupedanddoubled.zip](https://github.com/user-attachments/files/20148990/groupedanddoubled.zip)
[image]
[image]
[image]
Then you split the grouped meshes to parts.
[image]

RainerMtb, 2025-05-11:
[X2.zip](https://github.com/user-attachments/files/20150527/X2.zip)
[image]
[image]

I am not sure what you mean in your first post about grouping. I do indeed know how to combine objects to make them load correctly. What I am criticising is that some 3MF files are handled differently than others. Here I attached an example where that dialog I refer to shows up and the user can decide how to proceed. 
Both my examples contain objects on different heights. If you can explain the difference between those 3MF files please do so. From my point of view this dialog should also show up with my first example so we have consistent behavior. I find this different behavior unexpected and therefore buggy.

u89djt, 2025-05-11:
In X2, all the objects are in contact with each other, which makes it a good bet that they comprise a single object.

u89djt, 2025-05-11:
Nope, that's not it. If I create a grid of 9 cubes in Fusion 360 and separate one off and save that as a 3mf, the slicer offers to open it as a single object with parts at different heights. If I separate all the cubes a bit and save that as a 3mf, the slicer doesn't offer.
[offset.zip](https://github.com/user-attachments/files/20150843/offset.zip)
That suggests that the slicer doesn't think your threaded section is in contact with the head. 
Yeah, if I import your X1.3mf into fusion, shift the threaded body downward by 0.01mm and save that as a new 3mf, the slicer offers to open it all as a single object.
[X1m0p01.zip](https://github.com/user-attachments/files/20150849/X1m0p01.zip)
The question then is should the slicer have judged the two parts in your original file as touching.

u89djt, 2025-05-11:
Your original seems to be a perfect fit, so that does seem to be a bug. Perfectly contacting cubes are interpreted as contiguous, but your model isn't.
[image]

RainerMtb, 2025-05-11:
Thanks for investigating

TCPhelps, 2026-08-30:
I can confirm PrusaSlicer version 2.9.6 still has this bug. Prusaslicer scatters the designed supports within a Fusion-exported .3mf file to random locations - sometimes beside the part, sometimes "within" the body of the main part. The 3mf file is attached.

[image]

[edge protector.zip](https://github.com/user-attachments/files/31618998/edge.protector.zip)

u89djt, 2026-08-30:
(community triage) That's happening because the 3mf file contains four separate bodies, not four parts in a body. Bodies always lie in contact with the bed, so they're dropped to the bed when they arrive at any other height.
Here's a 3mf file of your bodies merged into parts of a single body created by opening your 3mf in Fusion, merging the mesh bodies, then exporting as a 3mf file:
[parts.zip](https://github.com/user-attachments/files/31619587/parts.zip)
That looks like this in the slicer:
[image]
I left the dropped bodies in the back there and right click->separate to parts the correct arrangement that came in as a single entity in the list. 
Saving your bodies as an stl file gives this, which also keep the parts in the spatial relationship you want and looks identical to that screenshot:
[samebutstl.zip](https://github.com/user-attachments/files/31619616/samebutstl.zip)
The 3mf file format provides more information than an stl, and that extra information in yours is that the four bodies are separate, so they get dropped to the bed. You can either merge them before you save to 3mf, or just save as stl.
Here's a 3mf file saved from the slicer with the correctly arranged parts and the printer I had as default:
[partsfromslicer.zip](https://github.com/user-attachments/files/31619678/partsfromslicer.zip)
You can drag and drop it into the slicer in your preferred environment and select "Import as 3D models only".
It seems like the information provided in your 3mf file is being respected by the slicer.

u89djt, 2026-08-30:
Correction: the data in the 3mf files that maintain the relative positions doesn't contain merged explicit parts, it's everything as one collection of triangles, just like the stl.

u89djt, 2026-08-30:
I've just tested the X2.3mf file from the comment in the earlier discussion and that again causes the slicer to offer to maintain the positions and open as parts rather than separate objects.
I see that in your 3mf there are gaps between the bodies, which I presume are intended to reduce adhesion of the predefined supports? I would guess that the slicer decides they're separate because of that and doesn't offer to open as parts of a single body.
[image]
If you move the faces of the pegs on the supports to meet the main object and save as a 3mf with those separate objects in it, the slicer offers to open it as a single object with multiple parts:
[nogaps.zip](https://github.com/user-attachments/files/31620782/nogaps.zip)
[image]
That's all I've got.

u89djt, 2026-09-01:
A quick workaround that doesn't need you to resave as a single mesh or edit the objects:
1) Add an object to the plate with right click->"add shape"->anything e.g. box.
2) Right click that object and "add part"->"load" and pick your 3mf file.
That loads the objects in the 3mf file at the coordinates in the 3mf file relative to the host object, but as a single part name in the list.
3) Right click that and "separate". Each part is in the list now.
4) Delete the added object from the list of parts and you're done except for the "sinking" entry because you've got negative z coords. Select and hit delete and the whole object will move up to lie on the plate.
Done.

u89djt, 2026-09-01:
I just fired up the 3.0.0 alpha and it imports objects at their coordinates in the file and leaves them there, so that's where you'll find the result you're looking for. The objects in your file are sunk below the bed and they stay there. When you import the 3mf, the objects are all selected. Right click the group on the bed and selected merge, then place on face on your supports (shortcut F or an icon at the top of the window). Obviously it's not going to change in 2.9.x now.

closed 2026-09-29 by prusaslicer-bot[bot]