prusa3d/PrusaSlicer #14547: Multiple objects are imported with an offset

Bergschrat1, 2025-05-11:
**Is your feature request related to a problem? Please describe.**
When I import multiple models that should fit together, one of the models is always shifted a little bit. When I import both models in Blender, FreeCAD or OrcaSlicer they are perfectly aligned.

[image]

**Describe the solution you'd like**
Models should not be imported with an offset.

**Describe how it would work**
I click "Add (Ctrl+I)", I select multiple models with shift and click open.
Result: the two models perfectly fit together just like in Blender, FreeCAD or Orca.

**Describe alternatives you've considered**
Currently I have to manually move the models so that they are aligned. But it's very hard to get this perfectly right on a small model and imperfections lead to problems later.

**Additional context**
Basically, I want to recreate the workflow that is described in this video with PrusaSlicer instead of OrcaSlicer: https://www.youtube.com/watch?v=zZp-CLhH1Ao
At 6:41 he shows the step in Orca that is not working in PrusaSlicer.

u89djt, 2025-05-11:
(Intrigued fellow user) You'll need to post the objects you're trying to import so someone can investigate. Put them in a zip file.

u89djt, 2025-05-11:
A bit of experimentation when I realised it could be simple: loading objects into my MK3S profile centres all the objects. I would guess that's what's happening for you. The supports surround the object so its centre is going to be similar but not identical to that of the object. It looks like an offset, whereas its location is established independently from the main object.
You have (at least) two workaround options.
1) Switch to a multimaterial printer profile and it will offer to import them as a parts of a single object with different tools assigned, and you can then just switch to the profile you need.
2) To achieve the same thing directly in your single tool profile, add a default box to the plate (right click->add), then right click that box and add part->load and select the object and support structure. They'll be imported maintaining their relative positions. You can then delete the box part from the group.

So they're not imported with an offset. They're imported individually centred on the bed. 
I agree that it could be good to have the option to place objects according to their intended location from independent files, but that's yet another check box.

While we're here, you can create resin supports in prusaslicer in a resin printer profile and export the "sliced" bed as an stl to then load into an FFM profile.

Bergschrat1, 2025-05-12:
@u89djt You are a slicer wizard! Option 2 worked like a charm! 🥳 
However, I think some option that can be enabled to have this behavior by default would be nice.
Here is a zip file with two models that should fit together but are offset when imported:
[base.zip](https://github.com/user-attachments/files/20156269/base.zip)

u89djt, 2025-05-13:
Ah, you've used the SLA supports in the slicer and it defines a different mesh for each element of the supports. Got you.
If you use your original support stl, that stays in one unit.

So a workaround that involves a lot of deleting in one case: if you want to print the supports and object with different settings that can't be changed with a modifier applied to a sub part, you can:
0) Separate to parts, not objects
1) Set the x and y coordinates of the assembled parts to something you won't forget
2) Copy and paste the assemblage and change the x and y coordinates of the new copy to the same as the original.
3) Delete the object from one and the supports from the other. 
It is a bit of a pain with the multi-part supports, but you can identify the object elements by clicking on the first part and holding the down cursor key till you see a bit of the object flash green and noting its part number(s).

Finding stuff like this before I'm in a hurry is handy for me :)

u89djt, 2025-05-13:
You can select all the parts by clicking the first, shift clicking the last, then control clicking the object parts to deselect them, then hit delete to take out all the support parts.
[SLAsupportspartitioned.zip](https://github.com/user-attachments/files/20195273/SLAsupportspartitioned.zip)
[image]