prusa3d/PrusaSlicer #14548: PrusaSlicer CLI on MacOS (ARM) segmentation faults on --merge flag

kevinvalk, 2025-05-11:
### Description of the bug

Whenever I use the command line to automatically export gcode I was having segmentation faults. I narrowed this down to the `--merge` flag and it seems to be model independent.  So a `/Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer --merge 3dbenchy.stl` triggers a segmentation fault. That said, adding a `--dont-arrange` allows for loading the GUI.

`/Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer --merge 3dbenchy.stl`
[image]

`/Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer --dont-arrange --merge 3dbenchy.stl`
[image]
[image]

_The plot does thickens, as adding an --export-gcode while also having --dont-arrange crashes on the more complex CLI invocation, but I have a hard time narrowing down what the exact combinations are..._

### Project file & How to reproduce

1. Download a model
2. Run the following command from CLI (on a Mac maybe ARM chip) `/Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer  --merge 3dbenchy.stl`
3. Observe segmentation fault

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Apple M1 Max - macOS Sequoia 15.4.1

### Printer model

MK4s

WayneKennedy, 2026-09-24:
Still reproducible on **2.9.6** on **Linux aarch64**, so this is not macOS-specific.

**Environment:** Raspberry Pi 5 (8 GB), Debian 12 bookworm arm64, Flathub `com.prusa3d.PrusaSlicer` 2.9.6 (`stable`, commit `49c07599…`, runtime `org.gnome.Platform/aarch64/50`), CLI only.

**Observed — two distinct crashes:**

1. `--merge` segfaults (exit 139) with **any** input, including a single known-good STL. At `--loglevel 5` the last lines are `TriangleMesh::repair() started` / `finished`, then the crash, before any slicing. Reproduced on several unrelated models.
   ```
   prusa-slicer --load profile.ini --merge --export-gcode --output out.gcode part.stl   # exit 139
   ```
2. `--merge --dont-arrange` with two models gets past that: it slices, **writes the complete G-code file**, and then still exits 139.
   ```
   prusa-slicer --load profile.ini --merge --dont-arrange --export-gcode --output out.gcode a.stl b.stl   # file written, exit 139
   ```

The same inputs slice fine without `--merge`, and `--merge` works on the Debian-packaged 2.5.0 on the same machine (provided every input sits on Z=0). So on ARM there seem to be two faults: one in arrange, one on the way out after G-code export.

Happy to test a 3.0 alpha build or collect a backtrace if that helps.

Chronic-Frolic, 2026-10-14:
Confirming on **Linux x86_64**, PrusaSlicer **2.9.6** (Fedora package, `PrusaSlicer-2.9.6+Fedora`; the Flathub 2.9.6 build also segfaulted on `--merge` for us), with the **default config** (empty `--datadir`) and two plain 20 mm cubes:

```
prusa-slicer --datadir /tmp/empty --export-gcode --merge a.stl b.stl -o out.gcode   # exit 139, no G-code
prusa-slicer --datadir /tmp/empty --info --merge a.stl b.stl                       # exit 139
prusa-slicer --datadir /tmp/empty --export-3mf --merge a.stl b.stl -o out.3mf      # exit 139
```

`--info` crashing too puts it in the merge/arrange step, before any slicing. In our `--loglevel 5` runs the last lines are the `TriangleMesh::repair()` messages; the run never reaches `BuildVolume bed_shape clasified`. A single model without `--merge` is fine, and so is `--merge --dont-arrange` with the cubes placed apart.

The second crash mentioned above (`--merge --dont-arrange`: G-code written, then exit 139) only happens for us when objects overlap. Two cubes both at the origin log `gcode path conflicts found between b.stl and a.stl` and then segfault. That matches the null `ConflictResult` write in `Print::export_gcode` analysed in #15973, so I think it's the same bug, triggered by plain overlap, not only by organic supports.

One more `--merge --dont-arrange` behaviour that bit us: an object that lies off the bed is dropped without a message, and the run exits 0 (a cube at 100,100 plus a cube at 300,300 on the default 200×200 bed gives G-code with one cube's filament). Without `--merge`, a lone off-bed object prints "All objects are outside of the print volume" but also exits 0, with no G-code.

Workaround that works on 2.9.6: place the parts yourself and pass one multi-object 3MF (build items with transforms), with `--dont-arrange` or without it. No `--merge` is needed, and the positions are kept.

<details><summary>Cube generator</summary>

```python
import struct
def cube(path, x0, y0, s):
    v = [(x0+dx*s, y0+dy*s, dz*s) for dx in (0,1) for dy in (0,1) for dz in (0,1)]
    f = [(0,2,3),(0,3,1),(4,5,7),(4,7,6),(0,1,5),(0,5,4),(2,6,7),(2,7,3),(0,4,6),(0,6,2),(1,3,7),(1,7,5)]
    with open(path, 'wb') as o:
        o.write(b'cube'.ljust(80, b' ') + struct.pack('<I', len(f)))
        for a, b, c in f:
            o.write(struct.pack('<3f', 0, 0, 0) + b''.join(struct.pack('<3f', *v[i]) for i in (a, b, c)) + b'\0\0')
cube('a.stl', 0, 0, 20); cube('b.stl', 0, 0, 20)          # both at the origin
cube('c.stl', 100, 100, 20); cube('d.stl', 140, 100, 20)  # apart, on the bed
cube('e.stl', 300, 300, 20)                               # off the bed
```
</details>

Jan-Soustruznik, 2026-11-09:
Hello @Chronic-Frolic. Thank you for your report. I believe this issue is no longer reproducible in the latest 3.0 release. Could you please check if it works correctly on your end now? Thank you, and have a great day