prusa3d/PrusaSlicer #14549: Cannot select PrusaLink for physical printer

JeffFaer, 2025-05-12:
### Description of the bug

The host type menu for my physical MK4S is missing PrusaLink and PrusaConnect as options.

I used to be able to use PrusaLink with this printer, but today after updating the configuration files for PrusaSlicer and adding the non-HF 0.4 nozzle to the printer, the option appears to have disappeared.

```
$ cat ~/.config/PrusaSlicer/physical_printer/mk4s.ini 
# generated by PrusaSlicer 2.8.1+linux-x64-GTK3 on 2024-12-15 at 22:25:20 UTC
host_type = prusalink
preset_name = Original Prusa MK4S HF0.4 nozzle;Original Prusa MK4S HF0.6 nozzle
preset_names = "Original Prusa MK4S HF0.4 nozzle";"Original Prusa MK4S HF0.6 nozzle"
print_host = prusa-mk4
printer_technology = FFF
printhost_apikey = <redacted>
printhost_authorization_type = key
printhost_cafile = 
printhost_password = 
printhost_port = 
printhost_ssl_ignore_revoke = 0
printhost_user = 
```

### Project file & How to reproduce

[bug_report.zip](https://github.com/user-attachments/files/20152372/bug_report.zip)

Printers > Cog icon to edit the physical printer > Host Type > PrusaLink and PrusaConnect are both missing

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2+flathub.org

### Operating system

Ubuntu 24.04.2

### Printer model

MK4S

pflugj, 2025-05-17:
Same problem here!
I also missed ALL previously installed Filament pre-sets

The Problem exist on:
Ubuntu 25.04 with PrusaSlicer 2.9.2+flathub.org
and 
Windows 11 Version 24H2 with PrusaSlicer 2.9.2

I tried to reinstall on Ubuntu with no luck :-(
also
running the configuration wizard does not restore the old settings

**Reverting the "Configuration "Snapshot to:** 
Active: 17/04/2025 10:30:24: Upgrade
PrusaSlicer version: 2.9.2
print: 0.20mm 100mms Linear Advance - Copy
filaments: Generic PLA
printer: MK2.5S @ Octopi * Original Prusa i3 MK2.5S
vendor: PrusaResearch, version: 2.2.4, min PrusaSlicer version: 2.9.1-alpha0
model: MK2.5S, variants: 0.4
model: MK3.9, variants: 0.4
model: MK3.9MMU3, variants: 0.4

Brought back the Filament and all the other setting :-)
This is working in Windows and Ubuntu :-)

Printer Models MK2.5S and MK3.9MMU3

cheers

The attached file is for the non working "Windows config folder"
[PrusaSlicer.zip](https://github.com/user-attachments/files/20263880/PrusaSlicer.zip)

JeffFaer, 2025-05-18:
Hmm this suddenly started working for me today... sorry for the noise!

closed 2025-05-18 by JeffFaer

Jan-Soustruznik, 2025-05-19:
Hello, @JeffFaer, That does sound like strange behavior if it's working now. If you have some time, please send us your config folder. We’ll try to check what happened. There might be an issue in our code, especially if you haven't made any updates to your printer profiles. Thank you, and have a nice day!

Hello, @pflugj, If the issue is still relevant, please create a new ticket for it. However, I’m not sure that the problem you reported is the same.

pflugj, 2025-05-24:
@Jan-Soustruznik

Today I tried "configuration update" on Kubuntu 25.04 and this time I did not loose the config :-)
.. so I am all good :-)

cheers and thanks for helping