prusa3d/PrusaSlicer #14559: Bridge generated in wrong direction

senicram, 2025-05-15:
### Description of the bug

Layer 76 is showing support.
[image]

Layer 77 is showing bridge generated in wrong direction. It should be in opposite direction into support.
I marked right direction of bridge which should be generated.
[image]

### Project file & How to reproduce

Insert support and generate nozzle movements.

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 10

### Printer model

MK4S

Jan-Soustruznik, 2025-05-16:
Hello, @senicram, Could you please share your project file(3MF) that demonstrates the issue? Thank you, and have a nice day!

senicram, 2025-05-18:
Here you are.
[karuzela_nowa_B_v2.3mf.zip](https://github.com/user-attachments/files/20274157/karuzela_nowa_B_v2.3mf.zip)

Jan-Soustruznik, 2025-05-19:
Hello, @senicram , Thank you for sharing your project file. However, I noticed that my layers 76/77 look different in provided file. Anyway, I believe I understand the issue you're referring to. The automatic algorithm checks multiple parameters and tries to choose the best bridge angle. I understand that in some cases, the automatic selection might choose the wrong angle. In such cases, you can manually set your own bridge angle under: **Printer Settings > Infill > Bridge Angle** _(The default value is "0", which means it's set to automatic.)_

I suggest closing this ticket. I will make a note of this case for future reference when we work on modifying the algorithm.

Thank you, have a nice day.

senicram, 2025-05-19:
Thanks for support and putting task modifying algo into Backlog ;)

RWL-69, 2025-09-13:
This is still an issue in a larger scale.  I was going to open a different but but will stick with this one for now.

When printing an object with a long open, but narrow span, the bridging algorithm uses the longest distance.  I tried both Classic and Aranchne perimeter settings.  Attached image is showing the bridge layer.  60mm long  Widest point is 19.8mm. 

[image]

Software should automatically select the shortest route for bridging.  I can select a bridging angle of 90 degrees and the bridging is correct for this model.   All other tests, bridging was the longest distance.

[Bridge_bug.zip](https://github.com/user-attachments/files/22312678/Bridge_bug.zip)

Attached file is the one I was having issues with.  

Load file, and slice.

sslupsky, 2026-08-18:
This issue has been around for a very long time and I do not understand why it is being ignored.  Sometimes the issue can be addressed using the bridging angle but in situations where you have multiple features in a polar (circular) arrangement this workaround does not work because it simply shifts the issue to one of the other features in the polar arrangement.  Please prioritize this issue (i see it is closed but it really should be reopened).