prusa3d/PrusaSlicer #14565: Add an infill painter for areas that need specific internal support

dementeddr, 2025-05-17:
I sometimes design objects where a very specific element of the object would benefit from 100% infill because it needs to be as strong as possible, but that level of infill would be extremely excessive for the rest of the print. 

It would be great if I could paint or drag-highlight or otherwise indicate that a specific portion of a print should be infilled to a different percentage than the rest of the object it is a part of.

Painting might not be adequate since that is more of a topological solution than a volumetric one. Maybe let us define spheres within the print bed space somehow, or rectangular prisms, or circle things with the mouse and project that area from the perspective of the camera.

neophyl, 2025-05-17:
This is already possible and is achieved through the use of modifiers.  For specific areas you could use a mesh modifier, for specific heights you can use a height range modifier etc.
https://help.prusa3d.com/article/modifiers_1767

dementeddr, 2025-05-20:
Ahh, gotcha. I didn't know about these. Thank you.

closed 2025-05-20 by dementeddr