prusa3d/PrusaSlicer #14570: Problems with the latest update

lumbarn6, 2025-05-18:
### Description of the bug

I can't slice any 3D object—basically, it tells me I don't have enough memory, and the thing is, before I updated it, I could do it just fine. I don't know what it's due to.

I got 16 GB of Ram

### Project file & How to reproduce

click on slice and finish xD

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

w10

### Printer model

elegoo neptune 2

Jan-Soustruznik, 2025-05-19:
Hello, @lumbarn6 , Could you please share your project file(3MF) that demonstrates the issue? Thank you, and have a nice day!

lumbarn6, 2025-05-21:
YES! 

How i can share u the 3MF?

lumbarn6, 2025-05-22:
well i think i solve it

TerPuc, 2025-06-02:
Hi @lumbarn6, assuming the issue has been resolved I am closing it. In case another issue occurs or the same one happens again, please feel free to open new ticket. Have a nice day!  😊