prusa3d/PrusaSlicer #14573: Building Eigen fails because URL_HASH is wrong

amatulic, 2025-05-18:
### Description of the bug

**Problem:** When following the instructions to build PrusaSlicer on Windows, the "Eigen" dependency does not build, resulting in a build error when trying to compile for Visual Studio.

The problem is the URL_HASH definition in `deps\+Eigen\Eigen.cmake`, which is just a 4-line file:
````
add_cmake_project(Eigen
    URL "https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.zip"
    URL_HASH SHA256=e09b89aae054e9778ee3f606192ee76d645eec82c402c01c648b1fe46b6b9857
)
````
**Solution:** Comment out the URL_HASH line and rebuild.

This problem was identified in issue #14384 (still open at the time of this writing).

### Project file & How to reproduce

1. Clone the repo afresh, with no dependencies included.
2. Follow the [Windows build directions](https://github.com/amatulic/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md).
3. In step 2A, the command `cmake --build .` runs through the dependency build process. It is easy to miss error messages because the process takes several hours.
4. When the dependency build is complete, go to the directory `deps\build\builds\Eigen` and notice that it's empty.
5. The next step, generating the Visual Studio project files, fails because Eigen is missing.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11

### Printer model

MK3S + MMU2S

sarrchri, 2025-05-21:
Duplicate of #14558 and fixed as of 6fd9846df131c671ac9f944c836536f04d354a53.
This issue should be closed. :)

dcorbin, 2025-05-30:
I've encountered this problem as well: 
```
-- Using src='https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.zip'
-- verifying file...
       file='/Users/dcorbin/projects/open-source/PrusaSlicer/deps/build/downloads/Eigen/eigen-3.3.7.zip'
-- SHA256 hash of
    /Users/dcorbin/projects/open-source/PrusaSlicer/deps/build/downloads/Eigen/eigen-3.3.7.zip
  does not match expected value
    expected: 'e09b89aae054e9778ee3f606192ee76d645eec82c402c01c648b1fe46b6b9857'
      actual: '4815118c085ff1d5a21f62218a3b2ac62555e9b8d7bacd5093892398e7a92c4b'
-- Hash mismatch, removing...
CMake Error at dep_Eigen-stamp/download-dep_Eigen.cmake:163 (message):
  Each download failed!
```
This is trying to build 2.9.2.

Testing by hand: 
```
dcorbin-mbp-m1~/Downloads % curl --output eigen-3.3.7.zip  https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2970k    0 2970k    0     0  4879k      0 --:--:-- --:--:-- --:--:-- 4878k
dcorbin-mbp-m1~/Downloads % sha256 eigen-3.3.7.zip
SHA256 (eigen-3.3.7.zip) = 4815118c085ff1d5a21f62218a3b2ac62555e9b8d7bacd5093892398e7a92c4b
dcorbin-mbp-m1~/Downloads %
```

dcorbin, 2025-05-30:
This commit has been made with a suggestion to cherry-pick it

https://github.com/prusa3d/PrusaSlicer/commit/6fd9846df131c671ac9f944c836536f04d354a53

HattonLe, 2025-06-01:
Issue also applies to building the source code "PrusaSlicer-version_2.9.2.tar.gz" in Ubuntu 22.04
Applying the 6fd9846 commit gets past this problem.