prusa3d/PrusaSlicer #14575: BUG: Solid infill in wrong positions

ulab, 2025-05-20:
### Description of the bug

I am getting weird solid infills in places where there should be no infill at all. I've seen this happen with objects that had errors, but PrusaSlicer doesn't report any on this object.

[image]

This also is dependent on  settings.

- default "0.20 mm STRUCTURAL" is fine
- change to 3 Perimeters has wrong infill
- change Infill to 16%, everthing is fine again
- change Infill to 20% and the extra infill happens at a different position

### Project file & How to reproduce

[6x5 Multibin Plate-issues.3mf.zip](https://github.com/user-attachments/files/20357757/6x5.Multibin.Plate-issues.3mf.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

macOS 15.5 (24F74)

### Printer model

Original Prusa XL - 5T Input Shaper 0.4 nozzle

Jan-Soustruznik, 2025-05-21:
Hello, @ulab , Thank you for your report. We will investigate it more thoroughly. I’ve created an internal issue (SPE-2768).

ulab, 2026-09-02:
I've just tested the original file in PS3 and on first look it seems to work better now. Will close this until I can reproduce it again in PS3.

closed 2026-09-02 by ulab