prusa3d/PrusaSlicer #14576: Search settings strange behaviour

revaip, 2025-05-21:
### Description of the bug

When I try to search for XY, it cannot find the shrinkage compensation. Maybe not a big problem, but could involve other strings, too.

https://github.com/user-attachments/assets/753bd991-8fe4-4698-ae50-63ed8ffbd5a8

### Project file & How to reproduce

works with empty project, newly installed slicer, and prevously created projects, too, so it seems independent.

[search settings bug.zip](https://github.com/user-attachments/files/20361353/search.settings.bug.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

v2.9.2 win64

### Operating system

Win11 24H2

### Printer model

MK3S, MK4

u89djt, 2025-05-21:
(From a fellow user) It's the first hit in the list when you've typed xy:
[image]

revaip, 2025-05-21:
> (From a fellow user) It's the first hit in the list 
No, that's different: size compensation is in a fix dimension (like 0,1mm) regardless of the size of the model, while shrinkage compensation is in %: 0,3% for 10mm is 10,03mm, while for 150mm 150,45mm

u89djt, 2025-05-21:
Aha, thank you!
[image]

u89djt, 2025-05-21:
Happens for "z", too
[image]
Add a space after the z in the box to get more hits but not the compensation
[image]
Deciding how to avoid returning every word containing a z must be fairly tortuous. The devs have my sympathy :)

revaip, 2025-05-21:
It's easy to search for shrinkage instead of xy. I don't have any problem with it, just wanted to report that the search box cannot find this particular setting. It must be just some string usage problem (end of string calculaton or something similar).

u89djt, 2025-05-21:
Yep, gathering similar issues and related commentary in one place can be handy.

closed 2026-09-29 by prusaslicer-bot[bot]