prusa3d/PrusaSlicer #14581: Infill Bug (Prusa Slicer Version 2.9.2)

juckendesAuge, 2025-05-22:
### Description of the bug

MK3.5S with Revo6 Hotend
0.25mm Nozzle (because the font was not printed well with a 0.4mm nozzle)
100% Infill (Because it is a movable part and is often folded in and out)
Infill bug on layer 8 (1.04mm) see picture.

### Project file & How to reproduce

[Rummikub_Stand_Support.3mf.zip](https://github.com/user-attachments/files/20396704/Rummikub_Stand_Support.3mf.zip)
[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2 (Flatpak)

### Operating system

Fedora release 42 (Adams)

### Printer model

Prusa MK3.5S

u89djt, 2025-05-23:
(Fellow user taking an interest) That isn't a bug per se, it's a phase of preparation for perimeters on top of infill. In this case, we can see that it isn't theoretically necessary, because you just want solid, quick laydown of plastic. You would imagine it being possible to construct a test that would stop it happening, but the risks and side effects might be prohibitive in context. I don't know.
The slicer doesn't comprehend the whole model, it just knows what's above and below, and either side within a layer. One phase of preparation that isn't performed for your model is creation of a reliable bridging environment to create a good platform for whatever's going to happen above. If you scale up your model and decrease the infill percentage, you can watch it do the preparatory work in successive layers. The logical tests it has to use just happen to leave you with a patch of different infill to what you wanted.
[twosizesRummikub_Stand_Support.zip](https://github.com/user-attachments/files/20408473/twosizesRummikub_Stand_Support.zip)
[image]
This is the equivalent layer to what you're seeing in your project:
[image]
If you try to adjust the settings to give you a solid body without switchinig infill types, you'll see that you can get very close, but it still performs all the tests, and in a couple of spots on one layer, it concludes that the next layer up could do something beneficial. However in this case, the result is unfortunate, as a brief perimeter extrusion occurs over a gap, so that might not look great. That's the closest to a bug I've found in this so far.
[fewertopRummikub_Stand_Support.zip](https://github.com/user-attachments/files/20408538/fewertopRummikub_Stand_Support.zip)
[image]

u89djt, 2025-05-23:
Someone just posted another example of this scenario, so that suggests to me that I might not have seen that this didn't happen previously; that it was introduced recently somehow. I'll stop speculating and hope a dev chimes in. I wrote this on that other post https://github.com/prusa3d/PrusaSlicer/issues/14582:
"(From a fellow user) Hi there, you're experiencing the same phenomenon as the issue immediately before yours. Have a look here: https://github.com/prusa3d/PrusaSlicer/issues/14581
I've had a go at explaining why it happens.
My guess is that this is a thoroughly known case because there's a lot of text on flat surfaces, and coming up with a safe exception that won't break things further down the line is somewhere lower down the priority list because the prints come out fine."

neophyl, 2025-05-23:
Its done this for ages.  Theres also a bunch of these on the forum too, like this one https://forum.prusa3d.com/forum/prusaslicer/slicer-adding-logo-into-middle-layers/  where the original poster has worked out what its doing after making his initial post.

Theres also many more here with the same basic problem https://github.com/prusa3d/PrusaSlicer/issues/12260  
or the one I always remember is this one https://github.com/prusa3d/PrusaSlicer/issues/8818

u89djt, 2025-05-23:
Is there an established workaround for situations in which it leads to poor quailty results or slows things down unacceptably and unnecessarily?

juckendesAuge, 2025-05-24:
i often use transparent filament. you can sometimes see the artifacts.
But then I use gyroid for the infill. it also looks very good. :)