prusa3d/PrusaSlicer #14582: Issue with Slicing 3D Object in PrusaSlicer

Helwings, 2025-05-22:
### Description of the bug

I have created a 3D object with a thickness of 4mm using SolidWorks. On one side, I added text and performed an extruded cut on the text. I then imported the object into PrusaSlicer.

When I place the object face-down on the table and initiate slicing at 1mm, I notice that the slicer adds a layer with the same text. However, in theory, there should be no text on this layer.

I have checked the model in Simplify3D, and no text appears there.

Could you please assist me with this issue?

Best regards,

### Project file & How to reproduce

[tableau pickleball_2_v1354.zip](https://github.com/user-attachments/files/20401560/tableau.pickleball_2_v1354.zip)

[image]

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

windows 10

### Printer model

???

Helwings, 2025-05-23:
I understand your point of view. It doesn't change the final piece, however the printing time is much longer.

u89djt, 2025-05-23:
Absolutely.

Jan-Soustruznik, 2025-06-23:
Closed as duplicate of #14581