prusa3d/PrusaSlicer #14584: Filament System Presets Disappear

LS1Inferno, 2025-05-23:
### Description of the bug

Running MK3S+ with 0.06 nozzle presets. 

I am trying to run multiple filaments with one extruder, going into G code after and forcing a manual filament change. (that would be a nice feature to be able to use m600 instead of T0, T1, etc)

The system preset filaments disappear when I select more than one extruder, when i select multi material single extruder all of the presets disappear and can not be re added through any means. 

This makes it impossible to me to use different filaments for the prints. 

### Project file & How to reproduce

When select under printers -> General -> Extruders with more than 1, the select multiple material single extruder, 
all of the system preset filaments are gone and none other can be added. 

[Filament Disappearing.zip](https://github.com/user-attachments/files/20402612/Filament.Disappearing.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.2

### Operating system

Windows 10 Pro

### Printer model

MK3S+

rtyr, 2025-05-23:
That is intentional. You should not select "multi material single extruder" anyway, if you don't have MMU unit, just inrease number of extruders.

closed 2025-05-23 by rtyr

rtyr, 2025-05-24:
That option is meant for specific hardware, like MMU unit. It is not meant for "M600 hack" (and it would also cause blobs on tower if enabled). Yes, the name of the parameter may be confusing.

Some filaments may still disappear. You can see compatible condition on "dependencies tab".