prusa3d/PrusaSlicer #14585: Object too Large when it is not

SlickMick66, 2025-05-24:
### Description of the bug

I have a Cocoon create, but modified the printer to have double the length (400mm) I have this set in the slicer, always have. It has been a while since I printed something over 200mm long, but now when I put in a 205 x 100 x 6 piece in when calculating it now thinks it is too big (thinks the bed is 200 x 200), even though the graphic shows it fitting on the 200 x 400 bed that it shows? Does not make sense 

[image]

### Project file & How to reproduce

The issue is that it will not slice successfully

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 10

### Printer model

Homemade

u89djt, 2025-05-24:
(From a fellow user)
Save the 3mf project for that screenshot, zip it, and post that zip file so someone can investigate.

u89djt, 2025-05-24:
I'm running 2.9.2. Interestingly, if you swap the X and Y dimensions of the bed, it'll happily slice that object. Obviously, you need to rotate it through 90 degrees, and the result wouldn't actually fit your bed.
[image]

SlickMick66, 2025-05-24:
It works!!, how the hell did you work that out???

u89djt, 2025-05-24:
I don't think I found your solution, because to my understanding, your bed is 390 long, not 390 wide? I just tried to think of things in the neighbourhood of the problem and explored what could happen. At least one of us is wrong, so we keep searching :) Tell me what you're thinking.

SlickMick66, 2025-05-24:
Yea, I get it, it is broken - the origin, I think when I created the profile I used the Cocoon create profile and just extended it, I may have to create a new one..

u89djt, 2025-05-24:
Do report back with how it goes, even if it works.

TerPuc, 2025-05-26:
Hi @SlickMick66, thanks for reporting! I see you have sequential printing enabled. If you disable it, the error goes away. However, this shouldn't happen. We'll look into it.

Internal issue SPE 2770 has been created.

SlickMick66, 2025-05-26:
Legend - you are right - Thank you!!

codefaux, 2026-01-22:
Still an issue with 2.9.4 -- @TerPuc is that internal issue still being looked into?

dn768, 2026-09-21:
Please do not close this!!!