prusa3d/PrusaSlicer #14588: Unable to export STL with PS 2.9.0 to 2.9.2

ror1948, 2025-05-24:
### Description of the bug

Exporting STL creates an error and the created STL does not represent design.
Gcode exports with no error and gcode seems to be ok.

[image]
[image]

### Project file & How to reproduce

[DJI_Tx_cam_mountDV.zip](https://github.com/user-attachments/files/20427903/DJI_Tx_cam_mountDV.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

win10

### Printer model

I3 MK3S

u89djt, 2025-05-24:
(From a fellow user)
HI there, the slicer doesn't perform boolean operations at all. It's not an error with your model, it's just telling you that the negative volumes won't be exported. I agree that the implication of the wording is that there's a specific problem.
Currently, you need to convert the negative volumes to parts and then you can export the whole thing. You can then perform boolean operations in another package if you need to.
[export.zip](https://github.com/user-attachments/files/20428059/export.zip)
Maybe the error could say something more like:
"The exported mesh will contain only unmodified parts. The slicer does not implement boolean operations on meshes. To export a negative volume, change its type to 'part'."

ror1948, 2025-05-24:
ok but that does not give me the design.
I want the negative volume parts to be negative volume.
Once you convert the negative volume to parts the negative volume is gone.
Do not see how your solution is a solution.
Open up the gcode that is attached....it gives me the design.

thanks

Jan-Soustruznik, 2025-05-26:
Hello @ror1948, this is not a bug. It is the first implementation and comes with certain limitations. The algorithm may fail on non-manifold geometries (e.g., those containing self-intersection, etc). In that case, the user is notified, and only positive volumes are exported._ Please note that PrusaSlicer is not intended to be primary software for editing STL models. We have an internal ticket (SPE-2480) open for future development of this feature, but it is not a priority at the moment.

ror1948, 2025-05-29:
> The slicer can't perform those boolean operations for you. It never does that and doesn't have that function.
> It assesses the presence or absence of plastic by performing logic between the different volumes as it moves around a layer. It doesn't explicitly store the resulting shape per se.
> The design from the perspective of the slicer is the group of positive and negative volumes.
> If you want a single stl object that represents the sliced result, you'll have to use another package to make it. You'll export all the parts by converting the negative volumes into parts and perfoming the subtractions in something like meshlab.
> The 3mf file you provided contains that design as positive and negative volumes, and it loads into 2.9.2 and slices with no problem.

Ok but how do I import a Prusa project into meshlab?It looks like it wants to import STL but I can create a proper STL in PrusaSlicer.thanks