prusa3d/PrusaSlicer #14590: stop being a compatible profile asap

username-sucks, 2025-05-25:
I'm always frustrated when a profile being a compatible profile or not
when saving it of course it is or why would i save it ?
namely if i use a custom printer profile
its a pain having to go into edit & click the flag and search for profile i used last time
it is to annoying took a while to figure out

 Tom

neophyl, 2025-05-28:
I'm English and I can't really understand your sentence structure either.

The devs want a copy of your configuration folder so they can see exactly what your profiles contain.  You can then describe step by step so you and they have the same starting point to work from and they have the same problem and can then potentially fix it.  Cant fix a problem if you don't see the problem.

But profiles are hidden by the use of 2 mechanisms.  There's the new Profile Dependency that only shows the ones from the dependency tree or there's the older compatibility fields that have been in there since before it was Prusa Slicer, from Slic3er.

Its quite easy to make a print profile or filament profile compatible with ANY printer.  

I'll use the Prusa PLA one from a MK3+ as an example.

When selected, assuming you are in expert mode you can navigate to the Dependencies screen for the filament -

[image]

You can see that inherits settings from the Prusa ini.  It also has a compatible printers condition string.  If you wanted to use this with any printer then you would clear out the Compatible printers field 

[image]

and then you would click the Detach from system Preset button.

[image]

You would then save the profile with a new non system Name.  At that point they are completely your own.  Future updates to the base profiles from Prusa will not be reflected in them.  

[image]

And now even with the option to NOT show incompatible presets 

[image]

it can be used on any printer

[image]

Hope thats useful to you in creating your own profiles in future.

The same can also be done with Print profiles too.

Its the way I run ALL of my personal printers and profiles.

username-sucks, 2025-05-29:
hi
thanks thats fixed it profiles  in green don't have to looking all the time

thanks Tom

closed 2025-06-02 by TerPuc