prusa3d/PrusaSlicer #14596: Application Crash when using Emboss Tool

johnkray, 2025-05-28:
### Description of the bug

Prusa Slicer crashes very consistently and repeatably when using the Emboss tool.

### Project file & How to reproduce

I't not always consistent which action causes the crash, but it crashes every time before I'm finishes editing the embossed text. It usually crashed when checking the "use surface" or "per glyph" box. Also when trying to adjust the height of the text.

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

MacOS 15.4.1

### Printer model

Prusa Core One

3DBob99, 2025-05-29:
I have experienced this too. PS 2.9.2. MacOS 15.5.

So when using the Emboss Tool, I save my project very frequently.

3DBob99, 2025-05-29:
Hi Jan,

I’ll share this file. As far as reproduction steps, I don’t have any good way to repeatably  force a crash. When working on this model, I would make a change: perhaps the text or the emboss details, and “learned” that I had to save frequently to avoid losing many changes.

[ MAGA 50pct Fist 8 PROJECT.zip](https://github.com/user-attachments/files/20507768/MAGA.50pct.Fist.8.PROJECT.zip)

johnkray, 2025-05-30:
> Hello [@johnkray](https://github.com/johnkray), [@3DBob99](https://github.com/3DBob99), Could you please share your project file(3MF) or the specific model that causes the problem? Reproduction steps would also be very helpful. We’re unable to diagnose the issue without more data.

Unfortunately, the file is a client file under NDA, so I cannot share. I have had issues with application crashing in the past as well. The best I can do for you right now it that the text string was fairly long and it seem to most consistently crash when selecting the "use surface" or "per glyph" box.

MisterLl, 2025-06-17:
@Jan-Soustruznik I've experienced this issue as well. The emboss tool seems to work adding text to an object and wrapping it to the object by selecting "per glyph" in the Emboss Text menu. Selecting "use surface" is what causes the file to crash. This happens for me both when debossing the text into the object by using a negative volume, or by using a modifier to change the colour of the text with an mmu3. This issue doesn't seem to occur when rolling back to Prusaslicer 2.7.3.

*Note - have added a screen recording of the crash to the attached zip. In this instance, I had to push the button several times before it crashed. Until the recording, it had always crashed the first time I had pressed the button. 

[Issue #14596 - crash using emboss tool.zip](https://github.com/user-attachments/files/20779139/Issue.14596.-.crash.using.emboss.tool.zip)

Version of Prusaslicer - Version 2.9.2

Operating System - MacOS Sequoia 15.5

Printer model - MK4S + MMU3

MisterLl, 2026-09-02:
Can confirm this issue still exists in PrusaSlicer 3.0.0 alpha 11