prusa3d/PrusaSlicer #14602: PrusaSlicer 2.9.2: Saving password for "PrusaSlicer/PrusaAccount/tokens" failed

tpancoast, 2025-05-29:
### Description of the bug

After I log into PrusaSlicer using my email and password, I get an error dialog that says:

Saving password for "PrusaSlicer/PrusaAccount/tokens" failed: Not enough memory
resources are available to process this command ..

The login works, but I have to log in again next time. I have reinstalled PrusaSlicer normally and as Admin. I have also deleted the Prusa AppData folder and reinstalled. No change.

### Project file & How to reproduce

After I log into PrusaSlicer using my email and password, I get an error dialog that says:

Saving password for "PrusaSlicer/PrusaAccount/tokens" failed: Not enough memory
resources are available to process this command ..

The login works, but I have to log in again next time.

[Saving password failed.zip](https://github.com/user-attachments/files/20497278/Saving.password.failed.zip)
[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

Windows 11 Pro 24H2

### Printer model

Core One

tpancoast, 2025-05-29:
As a follow up, I connected my Prusa account to Google, tried to log in with that, and I got the same error message. 

I also just installed Prusa Slicer on my laptop and it works there. So, it appears to be a local config problem. As I said, I deleted the C:\Users\[User]\AppData\Local\PrusaSlicer and that didn't work.

In case it's relevant, I installed PrusaSlicer before I had my Core One because I wanted to see what filament brands were covered by supplied profiles. I cancelled the login during that initial install.

JoshBflo, 2025-05-31:
This just started happening to me within the last week. I don't think I installed a new version in that time. I also haven't tried a reinstall yet. I confirmed I have free memory and hard drive space. This happens even with a completely empty plate so it's not related to model complexity or anything like that taking resources.

tpancoast, 2025-06-14:
Update: Running PrusaSlicer as Admin works around the problem. I can log in without getting the error and it stays logged in. If I switch it back to a normal user, the problem returns.

In case it helps, here's my log file (Run as a normal user, not admin). I loaded "C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer-console.exe" --loglevel=5. Then I went immediately to login and captured the log after I got the error message.

I think I cleaned everything that needs to be sanitized and replaced them with "*****".

[2025-06-14 14:26:01.284516] [0x00008f60] [trace]   App updater default download path: "C:\Users\*****\Downloads"
[2025-06-14 14:26:01.285518] [0x00008f60] [debug]   full path: 17328539711037480854
[2025-06-14 14:26:01.285518] [0x00008f60] [info]    single instance: undefined. other params: C:\\Program Files\\Prusa3D\\PrusaSlicer\\prusa-slicer-console.exe;--loglevel=5;
[2025-06-14 14:26:01.285518] [0x00008f60] [debug]   init wx instance checker 17328539711037480854.lock C:\Users\*****\AppData\Roaming\PrusaSlicer\cache\
[2025-06-14 14:26:01.285518] [0x00008f60] [info]    Instance check: Another instance not found or single-instance not set.
[2025-06-14 14:26:01.288516] [0x00008f60] [trace]   System language detected (user locales and such): en_US
[2025-06-14 14:26:01.291515] [0x00008f60] [trace]   Best translation language detected (may be different from user locales): en
[2025-06-14 14:26:01.293518] [0x00008f60] [trace]   Switching wxLocales to en
[2025-06-14 14:26:01.300516] [0x00008f60] [info]    Checking if indices need to be installed from resources...
[2025-06-14 14:26:03.023517] [0x00008f60] [debug]   Slic3r::GUI::UserAccountCommunication::UserAccountCommunication access_token empty!
[2025-06-14 14:26:03.269515] [0x00008f60] [trace]   No 3DConnexion device detected
[2025-06-14 14:26:03.271516] [0x00008f60] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB WorkingSet: 417MB; PrivateBytes: 441MB; Pagefile(peak): 441MB(443MB)
[2025-06-14 14:26:03.389515] [0x00008f60] [debug]   BuildVolume bed_shape clasified as: Rectangle
[2025-06-14 14:26:03.391515] [0x00008f60] [debug]   TriangleMesh::repair() started
[2025-06-14 14:26:03.393515] [0x00008f60] [debug]   TriangleMesh::repair() finished
[2025-06-14 14:26:03.451517] [0x00008f60] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2025-06-14 14:26:03.463515] [0x00008f60] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2025-06-14 14:26:03.478515] [0x00008f60] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2025-06-14 14:26:03.732515] [0x00008f60] [trace]   '1;9;-21;0;0;0;400;0;0;0;1;0;0;0;0;Segoe UI' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(true), Encoding(-1),
[2025-06-14 14:26:03.736516] [0x00008f60] [trace]   '1;8;-19;0;0;0;400;0;0;0;1;0;0;2;32;Arial' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(-1),
[2025-06-14 14:26:03.741517] [0x00008f60] [trace]   '1;9;-21;0;0;0;400;1;0;0;1;0;0;2;16;Times New Roman' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(-1),
[2025-06-14 14:26:03.745515] [0x00008f60] [trace]   '1;9;-21;0;0;0;400;0;0;0;1;0;0;2;32;Arial' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(-1),
[2025-06-14 14:26:03.748515] [0x00008f60] [trace]   '1;10;-23;0;0;0;700;0;0;0;1;0;0;2;48;Courier New' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(true), IsUsingSizeInPixels(false), Encoding(-1),
[2025-06-14 14:26:03.762515] [0x00008f60] [trace]   '1;9;-21;0;0;0;400;0;0;0;1;0;0;0;0;Segoe UI'font descriptor string param of load_wxFont()
[2025-06-14 14:26:03.765516] [0x00008f60] [trace]   '1;9;-21;0;0;0;400;0;0;0;1;0;0;0;0;Segoe UI' wx string descriptor
[2025-06-14 14:26:03.767515] [0x00008f60] [trace]   loaded font is 'Segoe UI'.
[2025-06-14 14:26:03.769515] [0x00008f60] [trace]   '1;9;-21;0;0;0;400;0;0;0;1;0;0;0;0;Segoe UI' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(true), Encoding(-1),
[2025-06-14 14:26:04.417516] [0x00008f60] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2025-06-14 14:26:04.441518] [0x0000a2f0] [info]    Syncing configuration cache
[2025-06-14 14:26:04.443516] [0x0000a2f0] [info]    Get: `https://preset-repo-api.prusa3d.com/v1/repos/prusa-fff/vendor_indices.zip`
        -> `C:\Users\*****\AppData\Roaming\PrusaSlicer\cache\vendor_indices.zip`
        via tmp path `C:\Users\*****\AppData\Roaming\PrusaSlicer\cache\vendor_indices.zip.16332.download`
*   Trying 172.67.71.102:443...
*   Trying 172.67.71.102:443...
* Connected to files.prusa3d.com (172.67.71.102) port 443 (#0)
* schannel: ALPN, offering http/1.1
* Connected to preset-repo-api.prusa3d.com (172.67.71.102) port 443 (#0)
[... cut ...]
< Strict-Transport-Security: max-age=31536000; includeSubDomains
<
* Connection #0 to host connect.prusa3d.com left intact
[2025-06-14 14:26:24.465528] [0x000026b8] [trace]   UserActionGetWithEvent::perform on_complete
[2025-06-14 14:26:24.545716] [0x00008f60] [trace]   Prusa Connect printers message: {"printers": [{"uuid": "*********************************", "connect_state": "PRINTING", "tools": {"1": {"hardened": false, "material": "PLA", "high_flow": true, "nozzle_diameter": 0.4}}, "printer_model": "COREONE", "supported_printer_models": []}]}

Jan-Soustruznik, 2025-06-23:
Hello, @tpancoast , according to your update:

> Update: Running PrusaSlicer as Admin works around the problem. I can log in without getting the error and it stays logged in. If I switch it back to a normal user, the problem returns.

It seems that the issue may be related to permissions in your operating system, as it looks like PrusaSlicer doesn't have the necessary rights when run as a normal user. Could you please check your system’s settings or security configuration to see if there are any restrictions in place for the slicer application?

evanbrumley, 2025-06-24:
As an extra data point, I encountered this today. Had just upgraded to 2.9.2 and was logging in for the first time (just upgraded to a Core One so first time using online functionality). Same error message, which occurred after login whether running as administrator or not. Issue resolved after a full system reboot.

tpancoast, 2025-06-27:
And.... It seems to be gone. I started off by giving "Everyone" Full Control permissions on the "Prusa" folder. I ran the EXE directly and it worked. I unchecked the Admin checkbox in the shortcut, ran it that way, and it worked. I logged out of PrusaSlicer, restarted it, and logged back in and it worked. Maybe it had a one time write to the "Prusa" folder and now that it's done it doesn't need to do it anymore? Maybe MS updated something in the meantime? I don't know. Development/Troubleshooting is frustrating. But it's working.

KeirRice, 2025-12-19:
I had the same issue as OP.

On my system, the root cause was the windows Credential Manager getting so full that Prusa was unable to write to the vault.
The fix was to go to "Control Panel\User Accounts\Credential Manager" -> Windows Credentials -> Generic Credentials and remove some unneeded entries.

In my specific case it was the Minecraft Launcher & XBox Live apps that had 100s of stale entries.