prusa3d/PrusaSlicer #14606: temp settings

3d-Printers, 2025-05-30:
### Description of the bug

not sure if it was from the firmware upgrade (just upgraded to the latest 2 days ago)
I have a prusa Xl
in prusa slicer on extruder 5 I have PC-CF loaded
in the slicer I have the temps set to 
Hotend - 265 first layer and 260 for all others
Bed Temp - 80

when the print starts it always set the temps to:
hotend 270-275
bed      100-110

### Project file & How to reproduce

[Gantry Flange.zip](https://github.com/user-attachments/files/20523056/Gantry.Flange.zip)

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.2

### Operating system

windows 11

### Printer model

Prusa XL

neophyl, 2025-05-30:
I loaded your project in.  Its showing the Print and printer profile as modified from the base settings but the only filament its showing as modified is the Prusament PETG loaded into extruder 1

[image]

When I look at whats configured in there it is set to the temps you mention 

[image]

The others are the standard profiles.  You can only edit one filament at a time.  If you change filaments then you need to save your changes.  Its one of the current limitations of the slicer.  

I suspect that is what is happening.  
See https://github.com/prusa3d/PrusaSlicer/issues/14528

Jan-Soustruznik, 2025-06-02:
Hello @3d-Printers, Thank you for your feedback. Yes, you're right — this is caused by a limitation in the current implementation. We’re aware of it and plan to improve the profile feature in one of the upcoming releases.

I’m closing this ticket as a duplicate of #14528.

closed 2025-06-02 by Jan-Soustruznik