prusa3d/PrusaSlicer #14610: No modification modificator

yaris37, 2025-06-01:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when I've used the modificator and forgot to set any modification. 

**Describe the solution you'd like**
Is it possible to popup an warning before slicing that there is a modificator used without any modification?

Thank you

yaris37, 2025-06-02:
> Odesláno: pondělí 2. června 2025 10:02
> Komu: prusa3d/PrusaSlicer ***@***.***>
>
>
>
>
> Hello @yaris37 <https://github.com/yaris37> , I’m not sure if this feature has already been requested. We've received a lot of negative feedback regarding warning messages, so we’re trying to avoid adding more of them.
>
> Do you think the side panel isn’t sufficient for checking the modifiers settings?

Dobrý den

 

Už se mi několikrát stalo, že jsem mezi nastavením modifikátoru a tiskem musel někam odskočit, a když jsem se vrátil, myslel jsem, že je nastavení hotové, ale nebylo. Postranní panel jsem nezkontroloval. Vím, že tam ta informace je. Možná by stačila výraznější ikona u nenastaveného modifikátoru místo varování? Tahle chyba ale stojí peníze a dost času, protože tisk se musí opakovat. Možná, že v tomhle případě má varování opravdu smysl.

 

S pozdravem

Jaroslav Křenek

kubispe1, 2025-06-04:
Ahoj @yaris37,

děkujeme za podnět a za upřesnění situace. Obecně se snažíme najít rovnováhu mezi užitečnými upozorněními a jejich množstvím – příliš mnoho výstrah často vede k tomu, že je uživatelé ignorují, nebo na ně dostáváme negativní zpětnou vazbu.

Z tohoto důvodu aktuálně neplánujeme implementovat výstrahu na prázdný modifikátor a tuto issue tímto uzavírám.

Varianta s výraznější ikonou nebo třeba ikonou s vykřičníkem v případě prázdného modifikátoru nám ale dává smysl. Tento nápad můžeme zvážit při úpravách UI.

Ještě jednou děkujeme za zpětnou vazbu!

S pozdravem,
@kubispe1